w111liang222/lidar-slam-detection #1: where is the Tviz code？

liuxinren456852, 2023-04-26:


liuxinren456852, 2023-04-27:
Thank you for your contribution to the community

luameows, 2023-05-31:
Same question. Any plan to release tviz code? In our situation, we use apollo to build our project, but apollo has very week visualization-tools especially for pointclouds. Here we want to try your tviz-tools to connect apollo and webui, but do not konw how to upload these msg (such as points, odom, maps and so on) to tviz-module.

w111liang222, 2023-05-31:
> Same question. Any plan to release tviz code? In our situation, we use apollo to build our project, but apollo has very week visualization-tools especially for pointclouds. Here we want to try your [...]

Okay, I think we will upload the js source code of web in next week.
Currently, If you want to publish the message like pointcloud, you can refer the code at:

1. sensor_driver/ins_driver/src/ins_driver.cpp line 144, where publish the imu message.
2. sensor_driver/common_lib/cpp_utils/Utils.h, there are many function can be used to transfer data to messages.

w111liang222, 2023-06-01:
@liuxinren456852 @luameows 

Hi,
We have been uploaded the web ui source code to the github, you can refer the README to install and test it

liuxinren456852, 2023-06-01:
Thank you very much, this issue can be closed

closed 2023-06-01 by liuxinren456852