w111liang222/lidar-slam-detection #2: Unable to find the 'Merge Maps' option in the http://localhost/editor

hwh666, 2023-04-28:
When I open http://localhost/editor When there is no 'merge map' option under the file, I have already pulled the latest library. The following is the output log information at startup

root@dns:/home/znqc/work/lidar-slam-detection# tools/scripts/start_system.sh
run server python script
dpkg-query: no packages found matching nvidia-l4t-core
Running on board: IPC, machine: x86_64, jetpack: 20.04
2023-04-28 09:42:16,937[27174:27174][module_manager.py: 47] pipeline: Source connect to SLAM
2023-04-28 09:42:16,937[27174:27174][module_manager.py: 47] pipeline: SLAM connect to Sink
2023-04-28 09:42:16,937[27174:27174][module_manager.py: 71] start to setup module: SLAM
2023-04-28 09:42:16.937[27174:27182][localization.cpp:13] Localization: release static resources
2023-04-28 09:42:16.937[27174:27182][hdl_graph_slam_nodelet.cpp:122] releasing hdl graph nodelet...
2023-04-28 09:42:16,937[27174:27182][        slam.py: 51] SLAM use sensors: ['IMU', '0-RS-LiDAR-16', 'flir:0']
2023-04-28 09:42:16,937[27174:27174][module_manager.py: 73] setup module: SLAM, done
2023-04-28 09:42:16,937[27174:27174][module_manager.py: 71] start to setup module: Sink
iKD-Tree Multi thread started 
2023-04-28 09:42:16.942[27174:27182][prefiltering_nodelet.cpp:39] downsample: VOXELGRID 0.15
2023-04-28 09:42:16.942[27174:27182][prefiltering_nodelet.cpp:78] outlier_removal: NONE
2023-04-28 09:42:16.942[27174:27182][hdl_graph_slam_nodelet.cpp:51] initializing hdl graph nodelet...
2023-04-28 09:42:16.942[27174:27182][graph_slam.cpp:56] construct solver: lm_var
# Using CSparse poseDim -1 landMarkDim -1 blockordering 0
2023-04-28 09:42:16.942[27174:27182][loop_detector.hpp:50] initializing loop detection...
2023-04-28 09:42:16.942[27174:27182][registrations.cpp:57] registration: FAST_VGICP
2023-04-28 09:42:16.942[27174:27182][registrations.cpp:34] registration: FAST_GICP
2023-04-28 09:42:16,950[27188:27188][sink_template.py: 58] deamon UdpSink starts
2023-04-28 09:42:16,950[27174:27174][module_manager.py: 73] setup module: Sink, done
2023-04-28 09:42:16,950[27174:27174][module_manager.py: 71] start to setup module: Source
2023-04-28 09:42:16,954[27174:27174][player_data_manager.py:135] /home/znqc/work/slam_mapping_demo
2023-04-28 09:42:16,954[27174:27174][player_data_manager.py:136] player have 1942 frames, duration 194.099473 s
 * Serving Flask app 'module.source.player_data_manager'
 * Debug mode: off
2023-04-28 09:42:17,007[27174:27174][source_manager.py: 35] Source, start to setup: lidar
WARNING: This is a development server. Do not use it in a production deployment. Use a production WSGI server instead.
 * Running on http://127.0.0.1:38000
Press CTRL+C to quit
2023-04-28 09:42:17.009[27174:27174][lidar_driver_wrapper.cpp:34] lidar 0:RS-LiDAR-16 open success
2023-04-28 09:42:17.009[27174:27174][lidar_driver_wrapper.cpp:46] set lidar 0:RS-LiDAR-16 external parameter [0.0, 0.0, 0.0, 178.35, 0.0, 0.0]
2023-04-28 09:42:17.009[27174:27174][lidar_driver_wrapper.cpp:58] set lidar 0:RS-LiDAR-16 range filter [-200.0, -200.0, -50.0, 200.0, 200.0, 50.0]
2023-04-28 09:42:17.009[27174:27174][lidar_driver_wrapper.cpp:70] set lidar 0:RS-LiDAR-16 exclude [-1.0, -1.0, -2.5, 1.0, 1.0, 2.5]
2023-04-28 09:42:17,009[27174:27174][source_manager.py: 38] Source, setup: lidar, done
2023-04-28 09:42:17,010[27174:27174][source_manager.py: 35] Source, start to setup: camera
2023-04-28 09:42:17,015[27174:27174][camera_data_manager.py:207] camera flir:0, try cap: souphttpsrc timeout=0 location=http://127.0.0.1:38000/v1/camera/0 ! jpegparse ! jpegdec ! video/x-raw  ! videoscale ! video/x-raw,width=1024,height=768  ! videoconvert ! video/x-raw,width=1024,height=768,format=I420 ! tee name=t
    t. ! queue ! jpegenc ! shmsink socket-path=/tmp/camera_jpeg_0 sync=false
    t. ! queue ! appsink sync=false 
127.0.0.1 - - [28/Apr/2023 09:42:17] "GET /v1/camera/0 HTTP/1.1" 200 -
[ WARN:0@0.266] global /tmp/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (1405) open OpenCV | GStreamer warning: Cannot query video position: status=1, value=0, duration=-1
2023-04-28 09:42:17,041[27174:27174][camera_data_manager.py:217] camera flir:0 open success
2023-04-28 09:42:17,041[27174:27174][source_manager.py: 38] Source, setup: camera, done
2023-04-28 09:42:17,041[27174:27174][source_manager.py: 35] Source, start to setup: radar
2023-04-28 09:42:17,041[27174:27174][source_manager.py: 38] Source, setup: radar, done
2023-04-28 09:42:17,041[27174:27174][source_manager.py: 35] Source, start to setup: ins
2023-04-28 09:42:17.041[27174:27174][ins_driver_wrapper.cpp:19] ins open success
2023-04-28 09:42:17.041[27174:27174][ins_driver_wrapper.cpp:34] set ins external parameter [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
2023-04-28 09:42:17.042[27174:27174][ins_driver_wrapper.cpp:43] ins start success
2023-04-28 09:42:17.042[27174:27213][ins_driver.cpp:278] start ins udp receving, port 9888
2023-04-28 09:42:17.042[27174:27214][ins_driver.cpp:317] start ins com receving, device /dev/ttyIMU_GPS0
2023-04-28 09:42:17.042[27174:27215][ins_driver.cpp:378] start ins gps receving
2023-04-28 09:42:17,042[27174:27174][source_manager.py: 38] Source, setup: ins, done
2023-04-28 09:42:17,042[27174:27174][module_manager.py: 73] setup module: Source, done
2023-04-28 09:42:17.042[27174:27215][ins_driver.cpp:381] No GPSD running
2023-04-28 09:42:17,042[27174:27216][manager_template.py: 80] SLAM, FPS: 4.8
2023-04-28 09:42:17,042[27174:27218][data_manager_template.py: 57] Camera: flir:0: acquire_loop start
2023-04-28 09:42:17,042[27174:27174][  perception.py: 34] start success
[ WARN:1@0.268] global /tmp/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (1405) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
2023-04-28 09:42:17,044[27174:27217][manager_template.py: 80] Source, FPS: 4.7
2023-04-28 09:42:17.158[27174:27184][IMU_Processing.hpp:362] IMU Initial Done
2023-04-28 09:42:17.158[27174:27184][laserMapping.cpp:974] FastLio undistort points is empty
2023-04-28 09:42:18,044[27174:27219][module_manager.py:116] system status Running, running thread/process information:
2023-04-28 09:42:18,044[27174:27219][module_manager.py:120] Thread  27174:Perception daemon False ident 0x7f0b7fd44c00
2023-04-28 09:42:18,045[27174:27219][module_manager.py:120] Thread  27182:FastLIO    daemon True ident 0x7f0b5bec6700
2023-04-28 09:42:18,045[27174:27219][module_manager.py:120] Thread  27216:SLAM       daemon True ident 0x7f0b267fc700
2023-04-28 09:42:18,045[27174:27219][module_manager.py:120] Thread  27217:Source     daemon True ident 0x7f0b25ffb700
2023-04-28 09:42:18,045[27174:27219][module_manager.py:120] Thread  27218:Py-Cflir:0 daemon True ident 0x7f0b257fa700
2023-04-28 09:42:18,045[27174:27219][module_manager.py:120] Thread  27221:zerorpc    daemon True ident 0x7f0b0f7fe700
2023-04-28 09:42:18,045[27174:27219][module_manager.py:120] Thread  27226:Py-Jflir:0 daemon True ident 0x7f0b0ffff700
2023-04-28 09:42:21.471[27174:27187][hdl_graph_slam_nodelet.cpp:231] map update, average score 0.0, best score 0.19276465879244892, inlier ratio 0.7278611, dx 0.9071208, da 31.995329
2023-04-28 09:42:21.471[27174:27187][hdl_graph_slam_nodelet.cpp:236] map initializing, odometry not stable, resetting...
2023-04-28 09:42:22.372[27174:27187][hdl_graph_slam_nodelet.cpp:231] map update, average score 0.0, best score 0.12182340285386119, inlier ratio 0.84300596, dx 0.23566943, da 34.099247
2023-04-28 09:42:22.372[27174:27187][hdl_graph_slam_nodelet.cpp:234] map initializing, do not allow only rotation, resetting...
2023-04-28 09:42:22.972[27174:27187][hdl_graph_slam_nodelet.cpp:231] map update, average score 0.0, best score 0.08593199736664416, inlier ratio 0.86002994, dx 0.22220375, da 35.935818
2023-04-28 09:42:22.972[27174:27187][hdl_graph_slam_nodelet.cpp:234] map initializing, do not allow only rotation, resetting...
2023-04-28 09:42:26.076[27174:27187][hdl_graph_slam_nodelet.cpp:231] map update, average score 0.0, best score 0.08094863986928584, inlier ratio 0.91425645, dx 1.5321901, da 3.5760264
2023-04-28 09:42:26.076[27174:27187][hdl_graph_slam_nodelet.cpp:239] map initialize success
2023-04-28 09:42:28,148[27174:27219][module_manager.py:137] Process 27188:UdpSink    CPU   0.0%, Mem 132 MB
2023-04-28 09:42:28,249[27174:27219][module_manager.py:137] Process 27189:playback   CPU   0.0%, Mem 153 MB
2023-04-28 09:42:28,349[27174:27219][module_manager.py:137] Process 27174:Perception CPU   0.0%, Mem 272 MB
[... cut ...]
2023-04-28 09:42:32.079[27174:27187][hdl_graph_slam_nodelet.cpp:231] map update, average score 0.06410854959392537, best score 0.047140600227384206, inlier ratio 0.76601595, dx 1.2203963, da 34.50612
2023-04-28 09:42:32.079[27174:27187][hdl_graph_slam_nodelet.cpp:252] map update, keep finding better keyframe
2023-04-28 09:42:32.181[27174:27187][hdl_graph_slam_nodelet.cpp:231] map update, average score 0.06410854959392537, best score 0.047140600227384206, inlier ratio 0.76601595, dx 1.295662, da 36.685013
2023-04-28 09:42:32.181[27174:27187][hdl_graph_slam_nodelet.cpp:252] map update, keep finding better keyframe

w111liang222, 2023-04-28:
Hi,
Following the steps, you can use the "Merge Map":
1. In "Config" => "Device" page, change the work mode to "SLAM" and upload config.
2. In the detailed "SLAM" option which at the left column,  change to "Localization" mode and upload config.
3. Refrese the editor page, you can see the "Merge Map" option.

The reason is that we do not have the "editor" mode, so the "localization" mode is reused.

w111liang222, 2023-04-28:
Currently, LSD read/save the maps in the **defal
ult** path (LSD read the **/proc/mounts** and find the external disk which is mounted in /media and use it as the **default** path).

If you want to save/read the maps, you can simulate an external disk by:
'''
mount --bind /home/znqc/work/ /media/external/
'''
Then, you can see the external disk in the "home" page.

hwh666, 2023-04-28:
> Hi, Following the steps, you can find the "Merge Map":
> 
> 1. In "Config" => "Device" page, change the work mode to "SLAM" and upload config.
> 2. In the detailed "SLAM" setting which at the left column,  change to "Localization" mode and upload config.
> 3. Refresh the editor page, you can see the "Merge Map" option.
> 
> The reason is that we do not have the "editor" mode, so the "localization" mode is reused.

Thank you for your answer. The problem has been resolved

hwh666, 2023-04-28:
> Currently, LSD read/save the maps in the **default** path (LSD read the **/proc/mounts** and find the external disk which is mounted in /media and use it as the **default** path).
> 
> If you want to save/read the maps, you can simulate an external disk by: ''' mount --bind /home/znqc/work/ /media/external/ ''' Then, you can see the external disk in the "home" page.

Okay, thank you very much for your reminder. Thank you very much for your project

I found this problem before, I set a fixed folder **/map_data** locally and by modifying the has_extension_disk() function, it was able to do the read and save functions as well,Here is my code
```
def has_extension_disk():
    has_disk = True
    disk_name = '/home/znqc/work/map_data'
    with open('/proc/mounts') as f:
        for v in f:
            v = v.split()
            mount_name = v[1]
            if '/media' in mount_name:
                has_disk = True
                disk_name = mount_name
                break
    return has_disk, disk_name
```

I would like to ask if using a local path would be more convenient and compatible，Or what is the necessity of using a disk for reading and saving

w111liang222, 2023-04-29:
> Okay, thank you very much for your reminder. Thank you very much for your project
>
> I found this problem before, I set a fixed folder **/map_data** locally and by modifying the has_extension_disk() [...]

We use the external disk to store the data because the LSD can be running on both PC and embedded board like Xavier NX. These embedded boards normally don't have enough storage to save maps.

hwh666, 2023-05-05:
Hello, when I tested the "Map Merge" function, the two maps were stacked and not aligned

Here is how I tested it:
I split a dataset into two parts, build a map of both datasets, and then perform a map merge,These two maps use the same cfg.yaml file

Here is my log:
 ---------------------------------start merge Map
2023-05-05 18:39:52.622[433912:433912][map_loader.cpp:114] Localization: meta_file /home/znqc/work/map_data/lp_log/map/nofix2/graph/map_info.txt load graph map origin: 0.0, 0.0, 0.0, 0.0, 0.0, 0.0 mOriginIsSet: false
2023-05-05 18:39:52.634[433912:433912][map_loader.cpp:38] Localization: total 60 key frames
2023-05-05 18:39:52.634[433912:433912][hdl_graph_slam_nodelet.cpp:51] initializing hdl graph nodelet...
2023-05-05 18:39:52.634[433912:433912][graph_slam.cpp:56] construct solver: lm_var
#Using CSparse poseDim -1 landMarkDim -1 blockordering 0
2023-05-05 18:39:52.634[433912:433912][loop_detector.hpp:50] initializing loop detection...
2023-05-05 18:39:52.634[433912:433912][registrations.cpp:57] registration: FAST_VGICP
2023-05-05 18:39:52.634[433912:433912][registrations.cpp:34] registration: FAST_GICP
2023-05-05 18:39:52.634[433912:433912][graph_slam.cpp:355] loading pose graph...
2023-05-05 18:39:52.634[433912:433912][graph_slam.cpp:363] nodes  : 60
2023-05-05 18:39:52.634[433912:433912][graph_slam.cpp:364] edges  : 59
2023-05-05 18:39:52.634[433912:433912][robust_kernel_io.cpp:138] kernels: 0
2023-05-05 18:39:52.634[433912:433912][hdl_graph_slam_nodelet.cpp:1154] Graph: max_se3_vertex_id 65, max_normal_vertex_id 100000, max_edge_id 79
2023-05-05 18:39:52.635[433912:433912][map_loader.cpp:204] Localization: after resample to 20 key frames
2023-05-05 18:39:52.635[433912:433912][map_loader.cpp:225] Localization: after voxel filter to 16 key frames
2023-05-05 18:39:52.635[433912:433912][map_loader.cpp:75] Merged Map: total 126 key frames
---------------------------------merge Map finish
2023-05-05 18:39:52,645[433912:433912][module_manager.py: 95] system is paused
2023-05-05 18:39:52.759[433912:433912][hdl_graph_slam_nodelet.cpp:286] start wait for flush keyframe queue
2023-05-05 18:39:52.759[433912:433912][hdl_graph_slam_nodelet.cpp:309] flush keyframe queue done
2023-05-05 18:39:52.761[433912:433912][hdl_graph_slam_nodelet.cpp:286] start wait for flush keyframe queue
2023-05-05 18:39:52.761[433912:433912][hdl_graph_slam_nodelet.cpp:309] flush keyframe queue done

w111liang222, 2023-05-05:
> 

Hi,

These two maps are created at the origin of the coordinate (0, 0, 0). So, when you perform map merging, you need to find out the two connected vertices in the maps and add an edge by manually (you can refer [this](https://github.com/w111liang222/lidar-slam-detection/blob/main/docs/slam.md#%E6%89%8B%E5%8A%A8%E5%9B%9E%E7%8E%AF)). Then, the pose graph will be optimized automatically and you will see the full aligned map.

In outdoor environment, if the GPS sensor is avaliable, you can specify the origin of the two maps by the same latitude and longitude, then, these two map can be merged easily because they are in the same coordinate system.

w111liang222, 2023-05-05:
hold "CTRL" key and click the vetex to select it

hwh666, 2023-05-06:
> 

Thank you for your reply. After using manual loopback, it can be aligned