w111liang222/lidar-slam-detection #3: How to use rosbag to build a map in LSD

hwh666, 2023-05-09:
I use the **rosbag_proxy** tool to send rosbag messages to LSD,But it seems that it can only run in Acquisition mode, not in slam mode

w111liang222, 2023-05-09:
> I use the **rosbag_proxy** tool to send rosbag messages to LSD,But it seems that it can only run in Acquisition mode, not in slam mode

Hi,

Did you transfer the IMU data to LSD?  The IMU data is necessary for FastLIO mapping and the LiDAR-IMU extrinic paramerters are also needed.
If you transfer only the pointcloud to LSD, you can use the GICPM and make sure the IMU item in "sensor input table" is deleted.
attach some logs here will be better.

hwh666, 2023-05-14:
Hi,
Now,I can use rosbag_proxy tool takes rosbag as the sensor data input, and can successfully work in the Mapping mode  and when I work in Localization mode,the system can run for a period of time, but then the program will make an error.   

here is my system parameter
Image size:  640 * 480
Image ROS Topic : /image_test  (type: sensor_msgs/Image)
parameter in board_cfg_all.yaml
```
board:
  fusion:
    source:
    - camera
  inference_engine:
    deactive_image: true
    image_cfg_file: sensor_inference/cfgs/detection_image.yaml
    lidar_cfg_file: sensor_inference/cfgs/detection_lidar_large.yaml
camera:
- extrinsic_parameters:
  - 0
  - 0
  - 0
  - -90
  - 0
  - 90
  fisheye: false
  flip_method: 0
  intrinsic_parameters:
  - 617.971050917033
  - 616.44513152479
  - 327.710279392468
  - 253.976983707814
  - -0.28340811
  - 0.07395907
  - 0.00019359
  - 1.76187114e-05
  name: http://127.0.0.1:/image_test
  static_extrinsic_parameters:
  - -0.0653706043958664
  - -0.0912313312292099
  - -0.10446935892105103
  - -88.7276840209961
  - 3.580638885498047
  - 91.50584411621094
  stream:
    audio_name: Y02
    bitrate: 1000000
    enable_audio: false
    host: 192.168.1.101
    port: 8554
    sink: none
  undistortion: false
input:
  data_path: /home/znqc/work/demo_data
  mode: online
ins:
  device: /dev/ttyUSB0
  extrinsic_parameters:
  - 0
  - 0
  - 0
  - 0
  - 0
  - 0
  imu_extrinsic_parameters:
  - 0
  - 0
  - 0
  - 0
  - 0
  - 0
[... 1503 lines cut ...]
2023-05-13 14:47:48.160[1175376:1175423][lidar_driver_wrapper.cpp:138] 0:Custom No lidar data comes
2023-05-13 14:47:48.160[1175376:1175418][ins_driver.cpp:304] UDP: gps/host time diff 1792.511962890625 ms, check the time sync!
2023-05-13 14:47:48.160[1175376:1175418][ins_driver.cpp:304] UDP: gps/host time diff 1790.217041015625 ms, check the time sync!
2023-05-13 14:47:48.160[1175376:1175418][ins_driver.cpp:304] UDP: gps/host time diff 1783.219970703125 ms, check the time sync!
2023-05-13 14:47:48.160[1175376:1175418][ins_driver.cpp:304] UDP: gps/host time diff 1750.9541015625 ms, check the time sync!
2023-05-13 14:47:48.160[1175376:1175418][ins_driver.cpp:304] UDP: gps/host time diff 1750.93896484375 ms, check the time sync!
2023-05-13 14:47:48.160[1175376:1175418][ins_driver.cpp:304] UDP: gps/host time diff 1750.93701171875 ms, check the time sync!
2023-05-13 14:47:48.160[1175376:1175418][ins_driver.cpp:304] UDP: gps/host time diff 1749.156005859375 ms, check the time sync!
2023-05-13 14:47:48.160[1175376:1175418][ins_driver.cpp:304] UDP: gps/host time diff 1742.47412109375 ms, check the time sync!
2023-05-13 14:47:48.160[1175376:1175418][ins_driver.cpp:304] UDP: gps/host time diff 1733.905029296875 ms, check the time sync!
[Inferior 1 (process 1175376) detached]
2023-05-13 14:47:48.160[1175376:1175418][ins_driver.cpp:304] UDP: gps/host time diff 1729.530029296875 ms, check the time sync!
2023-05-13 14:47:48.160[1175376:1175418][ins_driver.cpp:304] UDP: gps/host time diff 1712.278076171875 ms, check the time sync!
...
2023-05-13 14:47:48.160[1175376:1175418][ins_driver.cpp:304] UDP: gps/host time diff 1480.248046875 ms, check the time sync!
2023-05-13 14:47:48,160[1175376:1175421][slam_manager.py:109] SLAM ouput is Empty
...
2023-05-13 14:47:48.161[1175376:1175418][ins_driver.cpp:304] UDP: gps/host time diff 1006.8837890625 ms, check the time sync!
2023-05-13 14:47:48.161[1175376:1175418][ins_driver.cpp:304] UDP: gps/host time diff 1001.282958984375 ms, check the time sync!
2023-05-13 14:47:48.161[1175376:1175418][ins_driver.cpp:164] Ins data is unordered, 1683960468161189, 1683960467166738
2023-05-13 14:47:48,161[1175376:1175423][data_manager_template.py: 66] Lidar: 0-Custom, acquire FPS: 5.2
2023-05-13 14:47:48,162[1175376:1175422][manager_template.py: 80] Source, FPS: 5.2
2023-05-13 14:47:48.163[1175376:1175422][ins_driver.cpp:254] No new ins data within 1 period
2023-05-13 14:47:48.164[1175376:1175422][ins_driver.cpp:254] No new ins data within 1 period
2023-05-13 14:47:48.165[1175376:1175422][ins_driver.cpp:254] No new ins data within 1 period

w111liang222, 2023-05-15:
Hi, 

From the log, I find out the program crashed at here (the visual odometry library 
for map colouration):

_#4 0x00007f30e6eaa85a in Image_frame::get_rgb(double&, double, int, Eigen::Matrix<double, 3, 1, 0, 3, 1>, Eigen::Matrix<double, 3, 1, 0, 3, 1>) () at /home/znqc/work/lidar-slam-detection/build/lib.linux-x86_64-3.8/libvisual_odometry.so
#5 0x00007f30e6e60396 in VisualOdometry::vio_photometric(StatesGroup&, Rgbmap_tracker&, std::shared_ptr<Image_frame>&) () at /home/znqc/work/lidar-slam-detection/build/lib.linux-x86_64-3.8/libvisual_odometry.so
#6 0x00007f30e6e61bc1 in VisualOdometry::processLoop() () at /home/znqc/work/lidar-slam-detection/build/lib.linux-x86_64-3.8/libvisual_odometry.so_

The index of image to get the rgb value maybe out of bound, then I check the extrinic parameter of the camera :

_camera:
- extrinsic_parameters:
  - 0
  - 0
  - 0
  - -90
  - 0
  - 90_

The parameter seems not to be calibrated. You can firstly calibrate the extrinic parameter with the lidar and try to run localization again, or turn off the map colouration if you don't need the rgb map.