w111liang222/lidar-slam-detection #5: How to visualize the mapping process

luameows, 2023-05-26:
Thx for ur great job.
I check the api of slam.h and slam.py but do not understand how do u visualize the update keyframe when mapping. Cause the odom of the keyframe will be update when backend finishs the optimization, I think you should continuely monitor the keyframe queue. But when I look around the api, the "update_odom" will deque all the keyframe from mKeyframeQueue. In such case, the mKeyframeQueue will be empty. So I think the   "update_odom" should only be run after the mapping is finished?

w111liang222, 2023-05-26:
Hi,

In "py::dict update_odom()", we call the **graph_update_odom** to check if the graph of backend was optimized and get the updated pose of each frame.

When you visit the preview page, the js code will call **def get_pose(self)** (in slam.py) every 3s to query the newest map.

luameows, 2023-05-26:
> Hi,
> 
> In "py::dict update_odom()", we call the **graph_update_odom** to check if the graph of backend was optimized and get the updated pose of each frame.
> 
> When you visit the preview page, the js code will call **def get_pose(self)** (in slam.py) every 3s to query the newest map.

But in **py::dict update_odom()**, the function call "getKeyFrame"(in slam.h), and this function will use **try_dequeue** to pop frame in **mKeyframeQueue ** till the queue is empty?

w111liang222, 2023-05-26:
> But in **py::dict update_odom()**, the function call "getKeyFrame"(in slam.h), and this function will use **try_dequeue** to pop frame in **mKeyframeQueue ** till the queue is empty?

Yes, the mKeyframeQueue is used to transfer the keyframe from cpp to python.

We maintain the map by **map_manager.py** for web query and visuailize.

closed 2023-05-26 by luameows