w111liang222/lidar-slam-detection #7: Can see pointclouds in the web UI with your dataset

lingbomeng, 2023-07-05:
Hi, I have run LCD with your dataset. I can visit the web UI at http://localhost:1234/, but can not see something like pointclouds in the web.

Is there something wrong with my configuration?

The running result is as follows:

```
$ tools/scripts/start_system.sh
run server python script
dpkg-query: no packages found matching nvidia-l4t-core
Running on board: IPC, machine: x86_64, jetpack: 20.04
2023-07-05 15:42:00.743[77059:77059][SystemUtils.cpp:34] Failed to setschedparam: Operation not permitted
[Errno 2] No such file or directory: '~/work/licence/licences'
2023-07-05 15:42:00.811[77061:77061][SystemUtils.cpp:34] Failed to setschedparam: Operation not permitted
2023-07-05 15:42:00,898[77061:77061][module_manager.py: 47] pipeline: Source connect to SLAM
2023-07-05 15:42:00,898[77061:77061][module_manager.py: 47] pipeline: SLAM connect to Sink
2023-07-05 15:42:00,898[77061:77061][module_manager.py: 71] start to setup module: SLAM
2023-07-05 15:42:00.898[77061:77067][SystemUtils.cpp:34] Failed to setschedparam: Operation not permitted
2023-07-05 15:42:00.898[77061:77067][localization.cpp:13] Localization: release static resources
2023-07-05 15:42:00.898[77061:77067][hdl_graph_slam_nodelet.cpp:123] releasing hdl graph nodelet...
2023-07-05 15:42:00,898[77061:77067][        slam.py: 51] SLAM use sensors: ['IMU', '1-Ouster-OS1-128']
2023-07-05 15:42:00,899[77061:77061][module_manager.py: 73] setup module: SLAM, done
2023-07-05 15:42:00,899[77061:77061][module_manager.py: 71] start to setup module: Sink
iKD-Tree Multi thread started 
2023-07-05 15:42:00,905[77068:77068][sink_template.py: 58] deamon UdpSink starts
2023-07-05 15:42:00.906[77061:77067][prefiltering_nodelet.cpp:39] downsample: VOXELGRID 0.2
2023-07-05 15:42:00.906[77068:77061][SystemUtils.cpp:34] Failed to setschedparam: Operation not permitted
2023-07-05 15:42:00.906[77061:77067][prefiltering_nodelet.cpp:78] outlier_removal: NONE
2023-07-05 15:42:00.906[77061:77067][hdl_graph_slam_nodelet.cpp:52] initializing hdl graph nodelet...
2023-07-05 15:42:00.906[77061:77067][graph_slam.cpp:56] construct solver: lm_var
SOLVER FACTORY WARNING: Unable to create solver lm_var
2023-07-05 15:42:00.906[77061:77067][graph_slam.cpp:63] error : failed to allocate solver!!
gn_var_cholmod       CHOLMOD 	Gauss-Newton: Cholesky solver using CHOLMOD (variable blocksize)
gn_fix3_2_cholmod    CHOLMOD 	Gauss-Newton: Cholesky solver using CHOLMOD (fixed blocksize)
gn_fix6_3_cholmod    CHOLMOD 	Gauss-Newton: Cholesky solver using CHOLMOD (fixed blocksize)
gn_fix7_3_cholmod    CHOLMOD 	Gauss-Newton: Cholesky solver using CHOLMOD (fixed blocksize)
lm_var_cholmod       CHOLMOD 	Levenberg: Cholesky solver using CHOLMOD (variable blocksize)
lm_fix3_2_cholmod    CHOLMOD 	Levenberg: Cholesky solver using CHOLMOD (fixed blocksize)
lm_fix6_3_cholmod    CHOLMOD 	Levenberg: Cholesky solver using CHOLMOD (fixed blocksize)
lm_fix7_3_cholmod    CHOLMOD 	Levenberg: Cholesky solver using CHOLMOD (fixed blocksize)2023-07-05 15:42:00,906[77061:77061][module_manager.py: 73] setup module: Sink, done

dl_var_cholmod       CHOLMOD 	Dogleg: Cholesky solver using CHOLMOD (variable blocksize)
gn_var_csparse       CSparse 	Gauss-Newton: Cholesky solver using CSparse (variable blocksize)
gn_fix3_2_csparse    CSparse 	Gauss-Newton: Cholesky solver using CSparse (fixed blocksize)
gn_fix6_3_csparse    CSparse 	Gauss-Newton: Cholesky solver using CSparse (fixed blocksize)
gn_fix7_3_csparse    CSparse 	Gauss-Newton: Cholesky solver using CSparse (fixed blocksize)
lm_var_csparse       CSparse 	Levenberg: Cholesky solver using CSparse (variable blocksize)
lm_fix3_2_csparse    CSparse 	Levenberg: Cholesky solver using CSparse (fixed blocksize)
lm_fix6_3_csparse    CSparse 	Levenberg: Cholesky solver using CSparse (fixed blocksize)
lm_fix7_3_csparse    CSparse 	Levenberg: Cholesky solver using CSparse (fixed blocksize)2023-07-05 15:42:00,906[77061:77061][module_manager.py: 71] start to setup module: Source

dl_var_csparse       CSparse 	Dogleg: Cholesky solver using CSparse (variable blocksize)
gn_pcg               PCG 	Gauss-Newton: PCG solver using block-Jacobi pre-conditioner (variable blocksize)
gn_pcg3_2            PCG 	Gauss-Newton: PCG solver using block-Jacobi pre-conditioner (fixed blocksize)
gn_pcg6_3            PCG 	Gauss-Newton: PCG solver using block-Jacobi pre-conditioner (fixed blocksize)
gn_pcg7_3            PCG 	Gauss-Newton: PCG solver using block-Jacobi pre-conditioner (fixed blocksize)
lm_pcg               PCG 	Levenberg: PCG solver using block-Jacobi pre-conditioner (variable blocksize)
lm_pcg3_2            PCG 	Levenberg: PCG solver using block-Jacobi pre-conditioner (fixed blocksize)
lm_pcg6_3            PCG 	Levenberg: PCG solver using block-Jacobi pre-conditioner (fixed blocksize)
lm_pcg7_3            PCG 	Levenberg: PCG solver using block-Jacobi pre-conditioner (fixed blocksize)
-------------
2023-07-05 15:42:00,910[77061:77061][player_data_manager.py:135] player have 201 frames, duration 19.971323 s
2023-07-05 15:42:01.007[77070:77061][SystemUtils.cpp:34] Failed to setschedparam: Operation not permitted
 * Serving Flask app 'module.source.player_data_manager' (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off
2023-07-05 15:42:01,009[77061:77061][source_manager.py: 35] Source, start to setup: lidar
WARNING: This is a development server. Do not use it in a production deployment. Use a production WSGI server instead.
[... 61 lines cut ...]
2023-07-05 15:42:01,173[77061:77118][data_manager_template.py: 57] Camera: 1: acquire_loop start
2023-07-05 15:42:01.176[77061:77118][SystemUtils.cpp:34] Failed to setschedparam: Operation not permitted
2023-07-05 15:42:01,174[77061:77119][data_manager_template.py: 57] Camera: 2: acquire_loop start
2023-07-05 15:42:01.176[77061:77119][SystemUtils.cpp:34] Failed to setschedparam: Operation not permitted
2023-07-05 15:42:01,176[77061:77120][data_manager_template.py: 57] Camera: 3: acquire_loop start
2023-07-05 15:42:01.176[77061:77120][SystemUtils.cpp:34] Failed to setschedparam: Operation not permitted
[ WARN:1] global ../modules/videoio/src/cap_gstreamer.cpp (935) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
[ WARN:2] global ../modules/videoio/src/cap_gstreamer.cpp (935) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
[ WARN:3] global ../modules/videoio/src/cap_gstreamer.cpp (935) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
[ WARN:4] global ../modules/videoio/src/cap_gstreamer.cpp (935) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
2023-07-05 15:42:01.178[77061:77136][SystemUtils.cpp:34] Failed to setschedparam: Operation not permitted
2023-07-05 15:42:01.178[77061:77137][SystemUtils.cpp:34] Failed to setschedparam: Operation not permitted
2023-07-05 15:42:01.178[77061:77138][SystemUtils.cpp:34] Failed to setschedparam: Operation not permitted
2023-07-05 15:42:01.178[77061:77139][SystemUtils.cpp:34] Failed to setschedparam: Operation not permitted
2023-07-05 15:42:01,178[77061:77115][manager_template.py: 80] Source, FPS: 1.8
2023-07-05 15:42:03,197[77061:77115][manager_template.py: 80] Source, FPS: 1.1
/!\ gevent_zeromq BUG /!\ catching up after missing event (RECV) /!\
/!\ gevent_zeromq BUG /!\ catching up after missing event (RECV) /!\
2023-07-05 15:42:05,217[77061:77115][manager_template.py: 80] Source, FPS: 0.8
2023-07-05 15:42:07,238[77061:77115][manager_template.py: 80] Source, FPS: 0.7
/!\ gevent_zeromq BUG /!\ catching up after missing event (RECV) /!\
2023-07-05 15:42:09,258[77061:77115][manager_template.py: 80] Source, FPS: 0.6
/!\ gevent_zeromq BUG /!\ catching up after missing event (RECV) /!\

```

w111liang222, 2023-07-05:
Hi,
In the log, the lm solver of g2o was failed to construct, maybe the version of the libg2o is mismatched.
Can you run LSD using the provided docker image to check if the problem caused by the system environment

lingbomeng, 2023-07-06:
Thank you! I'll try your suggestions.