w111liang222/lidar-slam-detection #8: Cannot see the Config UI

CharlieV5, 2023-07-22:
Hi, I'm trying to test your program following the instructions here : https://github.com/w111liang222/lidar-slam-detection/blob/main/docs/slam.md
When I launch the LSD and open localhost:config , I cannot change the setting about mapping & localization. I just see the waiting page, as the picture below
[image]
[image]

w111liang222, 2023-07-23:
Hi,
What's the gpu card you use? The program was blocked when import the slam module.

You can try the "15liangwang/auto-ipu" docker image which is the non-cuda version

CharlieV5, 2023-07-23:
The GPU is RTX2060. The NVidia drivers and cuda have been installed. Maybe I will try non-cuda version.

w111liang222, 2023-07-24:
The SM80+ GPU is required by libspconv and the RTX2060 is SM76, you can check it from https://developer.nvidia.com/cuda-gpus.
So, we recommend to use the non-cuda version to test the LSD.

CharlieV5, 2023-07-25:
> The SM80+ GPU is required by libspconv and the RTX2060 is SM76, you can check it from https://developer.nvidia.com/cuda-gpus. So, we recommend to use the non-cuda version to test the LSD.

Thank you! I'll try later.

CharlieV5, 2023-08-07:
Hi, @w111liang222 
trying to use non-cuda docker, but now failed to start the docker container:
[image]

w111liang222, 2023-08-08:
Sorry, I didn't encounter this problem, you can try sudo or google the error message

CharlieV5, 2023-08-08:
> Sorry, I didn't encounter this problem, did you remove the "--gpus all" when you start the container

With or without "--gpus all" are both tried. 
[image]

w111liang222, 2023-08-08:
It is like an issue of nvidia docker, this link maybe can help you
https://github.com/NVIDIA/nvidia-docker/issues/1711

RuiYangQuan, 2023-08-21:
> The SM80+ GPU is required by libspconv and the RTX2060 is SM76, you can check it from https://developer.nvidia.com/cuda-gpus. So, we recommend to use the non-cuda version to test the LSD.

Hi, I'm experiencing the same problem, when I open localhost:config GUI keeps loading without displaying, both cuda version and non-cuda version I've tried.

XinyuMeng, 2023-09-08:
Hello, would you like to ask if your problem has been solved? I would like to share your experience. Because I encountered a similar problem to yours, my GPU model is RTX3070, and the error interface is the same as yours. Looking forward to your reply

w111liang222, 2023-09-08:
Hello, this issue maybe caused by the host software environment, the details of the host that we test LSD are:

- Ubuntu18.04
- docker-compose/bionic,now 1.17.1-2 all
- nvidia-docker2/bionic 2.13.0-1 all
- GPU: RTX3090

You can check it if the version is mismatch

XinyuMeng, 2023-09-11:
Hello,below is my environment information. It seems that only the GPU version is inconsistent with yours. I would like to ask you, how should I solve this problem next? Are there any examples of successful running of your code on the GPU model RTX3070? Thanks again。

$ dpkg -l | grep nvidia-docker2
ii  nvidia-docker2                               2.13.0-1                                        all          nvidia-docker CLI wrapper
$ dpkg -l | grep docker-compose
ii  docker-compose                               1.17.1-2                                        all          Punctual, lightweight development environments using Docker
ii  docker-compose-plugin                        2.18.1-1~ubuntu.18.04~bionic                    amd64        Docker Compose (V2) plugin for the Docker CLI.
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.6 LTS
Release:	18.04
Codename:	bionic

XinyuMeng, 2023-09-11:
> Hello, this issue maybe caused by the host software environment, the details of the host that we test LSD are:
> 
> * Ubuntu18.04
> * docker-compose/bionic,now 1.17.1-2 all
> * nvidia-docker2/bionic 2.13.0-1 all
> * GPU: RTX3090
> 
> You can check it if the version is mismatch
Hello, below is my environment information. It seems that only the GPU version is inconsistent with yours. I would like to ask you, how should I solve this problem next? Are there any examples of successful running of your code on the GPU model RTX3070? Thanks again
(base) mxy@mxy-System-Product-Name:~/CODES$ dpkg -l | grep nvidia-docker2
ii  nvidia-docker2                               2.13.0-1                                        all          nvidia-docker CLI wrapper
(base) mxy@mxy-System-Product-Name:~/CODES$ dpkg -l | grep docker-compose
ii  docker-compose                               1.17.1-2                                        all          Punctual, lightweight development environments using Docker
ii  docker-compose-plugin                        2.18.1-1~ubuntu.18.04~bionic                    amd64        Docker Compose (V2) plugin for the Docker CLI.
(base) mxy@mxy-System-Product-Name:~/CODES$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.6 LTS
Release:	18.04
Codename:	bionic

XinyuMeng, 2023-09-11:
> Hello, this issue maybe caused by the host software environment, the details of the host that we test LSD are:
> 
> * Ubuntu18.04
> * docker-compose/bionic,now 1.17.1-2 all
> * nvidia-docker2/bionic 2.13.0-1 all
> * GPU: RTX3090
> 
> You can check it if the version is mismatch

Hello, below is my environment information. It seems that only the GPU version is inconsistent with yours. I would like to ask you, how should I solve this problem next? Are there any examples of successful running of your code on the GPU model RTX3070? Thanks again

- nvidia-docker2 2.13.0-1 all nvidia-docker CLI wrapper
- docker-compose 1.17.1-2  all  
- docker-compose-plugin  2.18.1-1~ubuntu.18.04~bionic amd64  
- Ubuntu 18.04.6 LTS

w111liang222, 2023-09-11:
> 

I don't have RTX3070, so I only test the lsd with RTK3090. But the RTX3090 is similar to 3070, lsd should be able to run on both.
Did you try the non-cuda docker image (15liangwang/auto-ipu) to check if the problem is caused by the GPU

XinyuMeng, 2023-09-11:
> 

Yes, I have tested the non-cuda docker image (15liangwang/auto-ipu) version and it can be displayed normally.

XinyuMeng, 2023-09-11:
> > > Hello, this issue maybe caused by the host software environment, the details of the host that we test LSD are:
> > > 
> > > * Ubuntu18.04
> > > * docker-compose/bionic,now 1.17.1-2 all
> > > * nvidia-docker2/bionic 2.13.0-1 all
> > > * GPU: RTX3090
> > > 
> > > You can check it if the version is mismatch
> > 
> > 
> > Hello, below is my environment information. It seems that only the GPU version is inconsistent with yours. I would like to ask you, how should I solve this problem next? Are there any examples of successful running of your code on the GPU model RTX3070? Thanks again
> > 
> > * nvidia-docker2 2.13.0-1 all nvidia-docker CLI wrapper
> > * docker-compose 1.17.1-2  all
> > * docker-compose-plugin  2.18.1-1~ubuntu.18.04~bionic amd64
> > * Ubuntu 18.04.6 LTS
> 
> I don't have RTX3070, so I only test the lsd with RTK3090. But the RTX3090 is similar to 3070, lsd should be able to run on both. Did you try the non-cuda docker image (15liangwang/auto-ipu) to check if the problem is caused by the GPU

Yes, I have tested the non-cuda docker image (15liangwang/auto-ipu) version and it can be displayed normally.