w111liang222/lidar-slam-detection #13: How to use other format datasets to test performance

XinyuMeng, 2023-09-12:
Thank you for open source this excellent work, it is worthy of my serious study. I noticed that the data you provided are all in python's .pkl format. If I want to test other SLAM datasets such as NCLT, do I have to process them into .pkl files first? Or do you have other ideas?Looking forward to your reply (*^▽^*)

w111liang222, 2023-09-12:
Hello,
LSD supports communication with ROS, you can refer this https://github.com/w111liang222/lidar-slam-detection/blob/main/tools/rosbag_proxy/README.md

The shortcoming of rosbag_proxy is that it can not transfer the timestamp of each points, the SLAM use the timestamp to undistort the points. so the ghost points will be visualized in the scenario with large rotation motion

XinyuMeng, 2023-09-12:
> Hello, LSD supports communication with ROS, you can refer this https://github.com/w111liang222/lidar-slam-detection/blob/main/tools/rosbag_proxy/README.md
>
> The shortcoming of rosbag_proxy is that it [...]

Thank you for your reply. I just saw the relevant content of ROS interaction in your Zhihu article and am trying to use it.

XinyuMeng, 2023-09-14:
Sorry to bother you again. I encountered some problems that I couldn't solve when I tried to communicate between LSD and ROS. I would like to ask you to give me some solutions. When I run the tools/scripts/start_system.sh script, the terminal error message is as follows:
2023-09-14 11:17:45,115[2622:2622][source_manager.py: 38] Source, setup: ins, done
2023-09-14 11:17:45.115[2622:2660][ins_driver.cpp:381] No GPSD running

2023-09-14 11:06:06,547[2354:2539][detect_manager.py:100] frame queue is Empty
2023-09-14 11:06:06.547[2354:2546][radar_driver_wrapper.cpp:86] 0:ARS408 No radar data comes
2023-09-14 11:06:06,547[2354:2546][data_manager_template.py: 66] Radar: 0-ARS408, acquire FPS: 1.0
2023-09-14 11:06:06,547[2354:2540][ data_merger.py: 49] Detect has no output
2023-09-14 11:06:06.559[2354:2545][lidar_driver.cpp:37] lidar 0-Custom, receive buffer wait for 59 ms
2023-09-14 11:06:06.599[2354:2545][lidar_driver.cpp:37] lidar 0-Custom, receive buffer wait for 40 ms
2023-09-14 11:06:06.659[2354:2545][lidar_driver.cpp:37] lidar 0-Custom, receive buffer wait for 59 ms
2023-09-14 11:06:06.699[2354:2545][lidar_driver.cpp:37] lidar 0-Custom, receive buffer wait for 40 ms
2023-09-14 11:06:06.759[2354:2545][lidar_driver.cpp:37] lidar 0-Custom, receive buffer wait for 59 ms
2023-09-14 11:06:06.799[2354:2545][lidar_driver.cpp:37] lidar 0-Custom, receive buffer wait for 40 ms
2023-09-14 11:06:06.859[2354:2545][lidar_driver.cpp:37] lidar 0-Custom, receive buffer wait for 59 ms
2023-09-14 11:17:45,123[2622:2665][  can_driver.py: 36] IPC CAN is not implemented
2023-09-14 11:26:12.506[2964:3011][lidar_driver.cpp:37] lidar 0-Custom, receive buffer wait for 40 ms
2023-09-14 11:26:12.566[2964:3011][lidar_driver.cpp:37] lidar 0-Custom, receive buffer wait for 59 ms
2023-09-14 11:26:12.606[2964:3011][lidar_driver.cpp:37] lidar 0-Custom, receive buffer wait for 40 ms
2023-09-14 11:26:12.665[2964:3011][lidar_driver.cpp:37] lidar 0-Custom, receive buffer wait for 59 ms
2023-09-14 11:26:12.706[2964:3011][lidar_driver.cpp:37] lidar 0-Custom, receive buffer wait for 40 ms
2023-09-14 11:26:12.765[2964:3011][lidar_driver.cpp:37] lidar 0-Custom, receive buffer wait for 59 ms
2023-09-14 11:26:12.806[2964:3011][lidar_driver.cpp:37] lidar 0-Custom, receive buffer wait for 40 ms
2023-09-14 11:26:12.865[2964:3011][lidar_driver.cpp:37] lidar 0-Custom, receive buffer wait for 59 ms
2023-09-14 11:26:12.906[2964:3011][lidar_driver.cpp:37] lidar 0-Custom, receive buffer wait for 40 ms
2023-09-14 11:26:12.965[2964:3011][lidar_driver.cpp:37] lidar 0-Custom, receive buffer wait for 59 ms
2023-09-14 11:26:13.006[2964:3011][lidar_driver.cpp:37] lidar 0-Custom, receive buffer wait for 40 ms
2023-09-14 11:26:13,010[2964:2964][   http_sink.py: 92] http query FPS: 1.9
2023-09-14 11:26:13,053[2964:3005][detect_manager.py:100] frame queue is Empty
2023-09-14 11:26:13,053[2964:3006][ data_merger.py: 49] Detect has no output

Cannot find device "can0"
2023-09-14 13:52:50,562[3385:3429][  can_driver.py: 25] can0 link set down
Cannot find device "can0"
2023-09-14 13:52:50,564[3385:3429][  can_driver.py: 32] can0 baud set to 500000
Cannot find device "can0"
2023-09-14 13:52:50,565[3385:3429][  can_driver.py: 20] can0 link set up
Cannot find device "can0"
2023-09-14 13:52:50,566[3385:3429][  can_driver.py: 20] can0 link set up
2023-09-14 13:52:50,566[3385:3429][  can_driver.py: 36] IPC CAN is not implemented
2023-09-14 13:52:50.566[3385:3429][radar_driver_wrapper.cpp:63] radar 0:ARS408 start success
2023-09-14 13:52:50.566[3385:3445][radar_driver.cpp:86] start radar ARS408 receving, can port can0
2023-09-14 13:52:50.567[3385:3445][radar_driver.cpp:95] radar ARS408, can port can0 can't fit into the device

XinyuMeng, 2023-09-18:
Sorry to bother you again. I tried to receive rostopic on the host (using the demo.bag you provided), but an error occurred.  I tried for a long time but could not find out the reason. I would like to trouble you to provide me with some solution ideas. The error message is as follows:
2023-09-18 15:57:33,362[2064:2064][camera_data_manager.py:207] camera http://127.0.0.1:imageflir0, try cap: souphttpsrc timeout=0 location=http://127.0.0.1: 17777/stream?topic=imageflir0 ! jpegparse ! jpegdec ! video/x-raw ! videoscale ! video/x-raw ! videoconvert ! video/x-raw,format=I420 ! tee name=t
     t. ! queue ! jpegenc ! shmsink socket-path=/tmp/camera_jpeg_0 sync=false
     t. ! queue ! appsink sync=false
2023-09-18 15:57:33,393[2064:2064][camera_data_manager.py:210] camera http://127.0.0.1:imageflir0, can not open
--------
Try connecting via curl in docker:
(base) mxy@mxy-System-Product-Name:~/LSD$ curl -v http://myhost:17777/stream?topic=imageflir0
*   Trying myhost:17777...
* Connected to myhost (myhost) port 17777 (#0)
> GET /stream?topic=imageflir0 HTTP/1.1
> Host: myhost:17777
> User-Agent: curl/7.82.0
> Accept: */*
> 
* Mark bundle as not supporting multiuse
* HTTP 1.0, assume close after body
< HTTP/1.0 404 Not Found #URL在服务器上不存在或未被识别
< Content-Length: 85
< Content-Type: text/html
< 
* Excess found in a read: excess = 156, size = 85, maxdownload = 85, bytecount = 0
* Closing connection 0
<html><head><title>Not Found</title></head><body><h1>404 Not Found</h1></body></html>(base)

--------
The communication between my docker and the host is normal, and the lidar data and detection results can be displayed on the LSD web page.
----------
The complete terminal debug information is as follows:
root@mxy-System-Product-Name:~/exchange/lidar-slam-detection# tools/scripts/start_system.sh
run server python script
dpkg-query: no packages found matching nvidia-l4t-core
Running on board: IPC, machine: x86_64, jetpack: 20.04
访问时间: 2023-09-18 15:57:33, 访问的IP: ::1
访问时间: 2023-09-18 15:57:33, 访问的IP: ::1
2023-09-18 15:57:33,341[2064:2064][       infer.py: 31] image object detection is deactived
2023-09-18 15:57:33,346[2064:2064][module_manager.py: 74] Initialized modules: ['Split', 'Detect', 'SLAM', 'Merge', 'Sink']
2023-09-18 15:57:33,346[2064:2064][module_manager.py: 61] pipeline: Source connect to Split
2023-09-18 15:57:33,346[2064:2064][module_manager.py: 61] pipeline: Split connect to Detect
2023-09-18 15:57:33,346[2064:2064][module_manager.py: 61] pipeline: Detect connect to SLAM
2023-09-18 15:57:33,346[2064:2064][module_manager.py: 61] pipeline: SLAM connect to Merge
2023-09-18 15:57:33,346[2064:2064][module_manager.py: 61] pipeline: Merge connect to Sink
2023-09-18 15:57:33,346[2064:2064][module_manager.py: 89] start to setup module: Split
2023-09-18 15:57:33,346[2064:2064][module_manager.py: 91] setup module: Split, done
2023-09-18 15:57:33,346[2064:2064][module_manager.py: 89] start to setup module: Detect
2023-09-18 15:57:33,352[2064:2064][object_filter.py: 25] class mapping: {'vehicle': 1, 'pedestrian': 2, 'cyclist': 3, 'traffic_cone': 4}
2023-09-18 15:57:33,352[2064:2064][module_manager.py: 91] setup module: Detect, done
2023-09-18 15:57:33,352[2064:2064][module_manager.py: 89] start to setup module: SLAM
2023-09-18 15:57:33.353[2064:2075][localization.cpp:13] Localization: release static resources
2023-09-18 15:57:33.353[2064:2075][hdl_graph_slam_nodelet.cpp:123] releasing hdl graph nodelet...
2023-09-18 15:57:33,353[2064:2075][        slam.py: 52] SLAM use sensors: ['IMU', '1-Ouster-OS1-128']
2023-09-18 15:57:33,353[2064:2064][module_manager.py: 91] setup module: SLAM, done
2023-09-18 15:57:33,353[2064:2064][module_manager.py: 89] start to setup module: Merge
2023-09-18 15:57:33,353[2064:2064][module_manager.py: 91] setup module: Merge, done
2023-09-18 15:57:33,353[2064:2064][module_manager.py: 89] start to setup module: Sink
2023-09-18 15:57:33,356[2076:2076][sink_template.py: 58] deamon UdpSink starts
2023-09-18 15:57:33,356[2064:2064][module_manager.py: 91] setup module: Sink, done
2023-09-18 15:57:33,356[2064:2064][module_manager.py: 89] start to setup module: Source
2023-09-18 15:57:33,356[2064:2064][source_manager.py: 35] Source, start to setup: lidar
iKD-Tree Multi thread started 
2023-09-18 15:57:33.357[2064:2075][prefiltering_nodelet.cpp:39] downsample: VOXELGRID 0.3
2023-09-18 15:57:33.357[2064:2075][prefiltering_nodelet.cpp:78] outlier_removal: NONE
2023-09-18 15:57:33.357[2064:2075][hdl_graph_slam_nodelet.cpp:52] initializing hdl graph nodelet...
2023-09-18 15:57:33.357[2064:2075][graph_slam.cpp:56] construct solver: lm_var
# Using CSparse poseDim -1 landMarkDim -1 blockordering 0
2023-09-18 15:57:33.357[2064:2075][loop_detector.hpp:50] initializing loop detection...
2023-09-18 15:57:33.357[2064:2075][registrations.cpp:57] registration: FAST_VGICP
2023-09-18 15:57:33.357[2064:2075][registrations.cpp:34] registration: FAST_GICP
2023-09-18 15:57:33.358[2064:2064][lidar_driver_wrapper.cpp:34] lidar 0:Custom open success
2023-09-18 15:57:33.358[2064:2064][lidar_driver_wrapper.cpp:46] set lidar 0:Custom external parameter [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
[... 186 lines cut ...]
2023-09-18 15:57:35.467[2064:2113][lidar_driver.cpp:37] lidar 0-Custom, receive buffer wait for 59 ms
2023-09-18 15:57:35,467[2064:2111][data_manager_template.py: 64] Lidar: 0-Custom, acquire FPS: 10.0
2023-09-18 15:57:35,467[2064:2110][manager_template.py: 78] Source, FPS: 10.0
[... cut ...]
2023-09-18 15:57:35.566[2064:2113][lidar_driver.cpp:37] lidar 0-Custom, receive buffer wait for 34 ms
2023-09-18 15:57:35.594[2064:2104][SerialDriver.cpp:122] can not open device: /dev/ttyUSB0
2023-09-18 15:57:35,594[2064:2111][data_manager_template.py: 64] Lidar: 0-Custom, acquire FPS: 8.9
2023-09-18 15:57:35,595[2064:2110][manager_template.py: 78] Source, FPS: 8.9
2023-09-18 15:57:35,595[2064:2106][  infer_base.py: 76] lidarDet, prepare_data time is: 0.0 ms
2023-09-18 15:57:35.607[2064:2113][lidar_driver.cpp:37] lidar 0-Custom, receive buffer wait for 39 ms

XinyuMeng, 2023-09-18:
The lsd webpage is as follows：
[image]

XinyuMeng, 2023-09-19:
> > The lsd webpage is as follows： [image]
> 
> Hello, Do you mean the image can not be received by lsd? You can try to launch the rosbag proxy firstly, then run the lsd

Yes, LSD cannot receive image data. I tried the method you provided, but it doesn't seem to work. Are there any other solutions? Thanks

XinyuMeng, 2023-09-19:
I finally solved this problem, just change the name: http://127.0.0.1:imageflir0 in the configuration file to name: http://127.0.0.1:/imageflir0 😅 
[image]

XinyuMeng, 2023-09-27:
> > I finally solved this problem, just change the name: http://127.0.0.1:imageflir0 in the configuration file to name: http://127.0.0.1:/imageflir0 😅 [image]
> 
> Sorry, I didn't carefully check the camera url in the config😅

As shown in the picture above, In the communication mode between lsd and ros, the web preview interface only has the current point cloud information and 3D detection information, and does not display the complete map. i would like to know how to display the complete map in the web interface and how to add other attributes such as the speed information of each 3D object. Look forward to your answers.

w111liang222, 2023-09-27:
@XinyuMeng 
The provided rodbag is to test the rosbag proxy tool, here is the slam mapping demo data:  https://pan.baidu.com/s/1gEP5tYt2q3_omZD82OBDJg?pwd=g6u8 
If you want to test slam mapping with rosbag, you need to convert it to rosbag(using the provided pkl_to_rosbag tool) firstly.

XinyuMeng, 2023-09-28:
> @XinyuMeng The provided rodbag is to test the rosbag proxy tool, here is the slam mapping demo data: [https://pan.baidu.com/s/1gEP5tYt2q3_omZD82OBDJg?pwd=g6u8 [...]

Thank you very much for the data you provided. The slam mapping mode seems to be normal. What code should be modified if I want to display more information on the web page such as the speed of 3D objects. I am ashamed that I tried to read the source code you posted, but I am not very familiar with TVIZ and web-related knowledge and have no ideas, so I would like to ask you to give some tips. Thanks again♪(･ω･)ﾉ
[image]

w111liang222, 2023-09-28:
> Thank you very much for the data you provided. The slam mapping mode seems to be normal. What code should be modified if I want to display more information on the web page such as the speed of 3D [...]

[image]
you can enable the check box: “目标” - > "详细信息", the detail information of objects will be visualized, the text show the
" **id,x,y,length,width,heading,velocity_x,velocity_y,angle_rate,accelerate_x** "of each object.

lonlonago, 2024-10-25:
> > > 提供的 rodbag 是用来测试 rosbag 代理工具的，下面是 slam 映射的 demo 数据：[https://pan.baidu.com/s/1gEP5tYt2q3_omZD82OBDJg?pwd=g6u8 ](https://pan.baidu.com/s/1gEP5tYt2q3_omZD82OBDJg?pwd=g6u8%C2%A0) 如果你想用 rosbag 测试 slam 映射，需要先把它转换成 rosbag（使用提供的 pkl_to_rosbag 工具）。
> > 
> > 
> > 非常感谢您提供的数据。slam 映射模式似乎很正常。如果我想在网页上显示更多信息，例如 3D 对象的速度，应该修改什么代码。我很惭愧我尝试阅读您发布的源代码，但我对 TVIZ 和 Web 相关知识不是很熟悉，也没有什么想法，所以我想请您给出一些提示。再次♪感谢（・ω・）ノ[image]
> 
> ![图像](https://private-user-images.githubusercontent.com/14796737/271282786-69fbdddf-0c5b-4be0-b0e8-806c1f0a23e3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk4MjU1NjcsIm5iZiI6MTcyOTgyNTI2NywicGF0aCI6Ii8xNDc5NjczNy8yNzEyODI3ODYtNjlmYmRkZGYtMGM1Yi00YmUwLWIwZTgtODA2YzFmMGEyM2UzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDI1VDAzMDEwN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTVjNzIzODYzMWYzMGE1ZjNmMmM5MWRjOGE0OTc2ZTk2MzY3YTMyNDI2MDQ3ODAyMzU5NTM5MTE5NTI1YWI0MDUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.UHpqadZvW9RSXPaorE0BQnjJX0ixSBRF9EuQ92tFXDg)您可以启用复选框：“目标” - >“详细信息”，对象的详细信息将被可视化，文本显示每个对象的“**id，x，y，length，width，heading，velocity_x，velocity_y，angle_rate，accelerate_x** ”。

我想请问下，为什么没有显示出来目标预测的轨迹呢？但是看您的项目主页的动图的目标检测是有预测的轨迹出来的

lonlonago, 2024-10-25:
the rosbag_proxy only have lidar and imu topics for slam, Is it possible to use rosbag_proxy to include lidarand image for detection ?

w111liang222, 2024-10-25:
> the rosbag_proxy only have lidar and imu topics for slam, Is it possible to use rosbag_proxy to include lidar and image for detection ? how should I do it?

rosbag_proxy支持图像转发，原理是建立一个http server，lsd通过http请求获取图像，地库那个数据就转发了图像，单目相机检测只在v.1.x分支包含