w111liang222/lidar-slam-detection #14: The map editor can not load the map data

tops666, 2023-09-20:
hello ,thank you for your project. there was a problem with my test. the map editor can find the map flie. but  I get an error when I open the map file.

2023-09-20 10:08:08.633[723:832][map_loader.cpp:92] Localization: failed to load map meta data, /media/lp_log/map/odometry/map_info.txt
2023-09-20 10:08:08.633[723:832][localization.cpp:92] Map Loader: error to load map: /media/lp_log/map

w111liang222, 2023-09-20:
> hello ,thank you for your project. there was a problem with my test. the map editor can find the map flie. but I get an error when I open the map file.
>
> 2023-09-20 [...]

The map path should be 'lp_log/map/**xxx**', you should put the dumped map directory into 'lp_log/map/'

tops666, 2023-09-21:
> The map path should be 'lp_log/map/**xxx**', you should put the dumped map directory into 'lp_log/map/'

I do not understand where to set this directory.

w111liang222, 2023-09-21:
> I do not understand where to set this directory.

After running out the data, a pointcloud map has been built and we can use 'CTRL+ Left Click' to enter the map editor page, In 'FILE' menu, you can save the map to the disk (use 'mount --bind /root/exchange/ /media/external/' to simulate a external disk).
Then, you can open the saved in map editor.

tops666, 2023-09-22:
> After running out the data, a pointcloud map has been built and we can use 'CTRL+ Left Click' to enter the map editor page, In 'FILE' menu, you can save the map to the disk (use 'mount --bind [...]

thank you for your suggestion