w111liang222/lidar-slam-detection #16: nvidia embedded board build zerocm

bianhao3321, 2023-10-10:
在orin上编译项目，按照文档doc/nvidia/README.md配置完环境后，按照tools/scripts/install_dependency.sh流程python zcm/python/setup.py bdist_wheel过程无法通过，报错信息如下：[1/1] Cythonizing zcm/python/zerocm.pyx
performance hint: zcm/python/zerocm.pyx:81:5: Exception check on 'handler_cb' will always require the GIL to be acquired. Possible solutions:
	1. Declare the function as 'noexcept' if you control the definition and you're sure you don't want the function to raise exceptions.
	2. Use an 'int' return type on the function to allow an error code to be returned.
performance hint: zcm/python/zerocm.pyx:86:5: Exception check on 'handler_cb_raw' will always require the GIL to be acquired. Possible solutions:
	1. Declare the function as 'noexcept' if you control the definition and you're sure you don't want the function to raise exceptions.
	2. Use an 'int' return type on the function to allow an error code to be returned.
performance hint: zcm/python/zerocm.pyx:90:5: Exception check on 'handler_cb_deprecated' will always require the GIL to be acquired. Possible solutions:
	1. Declare the function as 'noexcept' if you control the definition and you're sure you don't want the function to raise exceptions.
	2. Use an 'int' return type on the function to allow an error code to be returned.
performance hint: zcm/python/zerocm.pyx:95:5: Exception check on 'handler_cb_raw_deprecated' will always require the GIL to be acquired. Possible solutions:
	1. Declare the function as 'noexcept' if you control the definition and you're sure you don't want the function to raise exceptions.
	2. Use an 'int' return type on the function to allow an error code to be returned.

Error compiling Cython file:
------------------------------------------------------------
...
        subs.handler = handler
        subs.msgtype = None
        sig = signature(handler)
        selected_handler_cb = handler_cb_raw_deprecated if len(sig.parameters) == 2 else handler_cb_raw
        while True:
            subs.sub = zcm_try_subscribe(self.zcm, channel.encode('utf-8'), selected_handler_cb, <void*> subs)
                                                                            ^
------------------------------------------------------------

zcm/python/zerocm.pyx:124:76: Cannot assign type 'void (*)(const zcm_recv_buf_t *, const char *, void *) except * nogil' to 'zcm_msg_handler_t'. Exception values are incompatible. Suggest adding 'noexcept' to type 'void (const zcm_recv_buf_t *, const char *, void *) except * nogil'.

Error compiling Cython file:
------------------------------------------------------------
...
        subs.handler = handler
        subs.msgtype = msgtype
        sig = signature(handler)
        selected_handler_cb = handler_cb_deprecated if len(sig.parameters) == 2 else handler_cb
        while True:
            subs.sub = zcm_try_subscribe(self.zcm, channel.encode('utf-8'), selected_handler_cb, <void*> subs)
                                                                            ^
------------------------------------------------------------

zcm/python/zerocm.pyx:136:76: Cannot assign type 'void (*)(const zcm_recv_buf_t *, const char *, void *) except * nogil' to 'zcm_msg_handler_t'. Exception values are incompatible. Suggest adding 'noexcept' to type 'void (const zcm_recv_buf_t *, const char *, void *) except * nogil'.
Traceback (most recent call last):
  File "zcm/python/setup.py", line 21, in <module>
    ext_modules=cythonize([
  File "/usr/local/lib/python3.8/dist-packages/Cython/Build/Dependencies.py", line 1154, in cythonize
    cythonize_one(*args)
  File "/usr/local/lib/python3.8/dist-packages/Cython/Build/Dependencies.py", line 1321, in cythonize_one
    raise CompileError(None, pyx_file)
Cython.Compiler.Errors.CompileError: zcm/python/zerocm.pyx

bianhao3321, 2023-10-10:
修改zcm/python/zerocm.pyx，对报错提示的callback函数添加noexcept修饰符后问题已解决。

closed 2023-10-10 by bianhao3321

wuhu-auto, 2025-06-16:
> 在orin上编译项目，按照文档doc/nvidia/README.md配置完环境后，按照tools/scripts/install_dependency.sh流程python zcm/python/setup.py bdist_wheel过程无法通过，报错信息如下：[1/1] Cythonizing zcm/python/zerocm.pyx performance hint: zcm/python/zerocm.pyx:81:5: Exception check on 'handler_cb' will always require the GIL to be acquired. Possible solutions: 1. Declare the function as 'noexcept' if you control the definition and you're sure you don't want the function to raise exceptions. 2. Use an 'int' return type on the function to allow an error code to be returned. performance hint: zcm/python/zerocm.pyx:86:5: Exception check on 'handler_cb_raw' will always require the GIL to be acquired. Possible solutions: 1. Declare the function as 'noexcept' if you control the definition and you're sure you don't want the function to raise exceptions. 2. Use an 'int' return type on the function to allow an error code to be returned. performance hint: zcm/python/zerocm.pyx:90:5: Exception check on 'handler_cb_deprecated' will always require the GIL to be acquired. Possible solutions: 1. Declare the function as 'noexcept' if you control the definition and you're sure you don't want the function to raise exceptions. 2. Use an 'int' return type on the function to allow an error code to be returned. performance hint: zcm/python/zerocm.pyx:95:5: Exception check on 'handler_cb_raw_deprecated' will always require the GIL to be acquired. Possible solutions: 1. Declare the function as 'noexcept' if you control the definition and you're sure you don't want the function to raise exceptions. 2. Use an 'int' return type on the function to allow an error code to be returned.
> ## Error compiling Cython file:
> ## ...
> subs.handler = handler
> subs.msgtype = None
> sig = signature(handler)
> selected_handler_cb = handler_cb_raw_deprecated if len(sig.parameters) == 2 else handler_cb_raw
> while True:
> subs.sub = zcm_try_subscribe(self.zcm, channel.encode('utf-8'), selected_handler_cb, <void*> subs)
> ^
> 
> zcm/python/zerocm.pyx:124:76: Cannot assign type 'void (*)(const zcm_recv_buf_t *, const char *, void *) except * nogil' to 'zcm_msg_handler_t'. Exception values are incompatible. Suggest adding 'noexcept' to type 'void (const zcm_recv_buf_t *, const char *, void *) except * nogil'.
> ## Error compiling Cython file:
> ## ...
> subs.handler = handler
> subs.msgtype = msgtype
> sig = signature(handler)
> selected_handler_cb = handler_cb_deprecated if len(sig.parameters) == 2 else handler_cb
> while True:
> subs.sub = zcm_try_subscribe(self.zcm, channel.encode('utf-8'), selected_handler_cb, <void*> subs)
> ^
> 
> zcm/python/zerocm.pyx:136:76: Cannot assign type 'void (*)(const zcm_recv_buf_t *, const char *, void *) except * nogil' to 'zcm_msg_handler_t'. Exception values are incompatible. Suggest adding 'noexcept' to type 'void (const zcm_recv_buf_t *, const char *, void *) except * nogil'. Traceback (most recent call last): File "zcm/python/setup.py", line 21, in ext_modules=cythonize([ File "/usr/local/lib/python3.8/dist-packages/Cython/Build/Dependencies.py", line 1154, in cythonize cythonize_one(*args) File "/usr/local/lib/python3.8/dist-packages/Cython/Build/Dependencies.py", line 1321, in cythonize_one raise CompileError(None, pyx_file) Cython.Compiler.Errors.CompileError: zcm/python/zerocm.pyx

1.有doc/nvidia/README.md这个文件吗。2、执行“tools/scripts/install_dependency.sh”后报错，“Install ZeroZCM...
Setting top to                           : /home/nvidia/znqc/work/lidar-slam-detection/tools/build/resources/ZeroCM 
Setting out to                           : /home/nvidia/znqc/work/lidar-slam-detection/tools/build/resources/ZeroCM/build 
Checking for 'gcc' (C compiler)          : /usr/bin/gcc 
Checking for 'g++' (C++ compiler)        : /usr/bin/g++ 
Checking for program 'pkg-config'        : /usr/bin/pkg-config 
Checking for 'libzmq'                    : not found 
The configuration failed
(complete log in /home/nvidia/znqc/work/lidar-slam-detection/tools/build/resources/ZeroCM/build/config.log)
The project was not configured: run "waf configure" first!
The project was not configured: run "waf configure" first!
Traceback (most recent call last):
  File "zcm/python/setup.py", line 3, in <module>
    from Cython.Build import cythonize
ModuleNotFoundError: No module named 'Cython'
WARNING: Requirement 'dist/zerocm-1.1.5-cp38-cp38-linux_aarch64.whl' looks like a filename, but the file does not exist
Looking in indexes: https://pypi.mirrors.ustc.edu.cn/simple/
Processing ./dist/zerocm-1.1.5-cp38-cp38-linux_aarch64.whl
ERROR: Could not install packages due to an EnvironmentError: [Errno 2] No such file or directory: '/home/nvidia/znqc/work/lidar-slam-detection/tools/build/resources/ZeroCM/dist/zerocm-1.1.5-cp38-cp38-linux_aarch64.whl'

/home/nvidia/znqc/work/lidar-slam-detection
/sbin/ldconfig.real: Can't create temporary cache file /etc/ld.so.cache~: Permission denied
。你有什么建议吗”