w111liang222/lidar-slam-detection #18: How to test performance in a real-world scenario using ROS

sycoolboy, 2024-01-11:
Hello, your work is excellent. I would like to use ROS to test the object detection performance of this code in a real-world scenario. What should I do? I don't have a GPU, and I'm using the Helios lidar and RealSense camera. I'm using the non-CUDA version (auto-ipu) of the code. Docker is already able to receive information from the lidar and camera, but it seems that object detection is not running. Is it because I haven't calibrated it? Below is the terminal information.
[image]
[image]
[image]
[image]
[image]

w111liang222, 2024-01-12:
hi,
the master branch of LSD use the libspconv to   realize the sparse 3D conv for object detection, it is built on CUDA, so, the nvidia gpu is necessary.

However, you can test the pointpillar detection model on non-cuda device with the v1.4 tag version. The openvino is used to inference the onnx model on the CPU(very slow, FPS 2~3Hz)

w111liang222, 2024-01-12:
pointpillar model
https://github.com/w111liang222/lidar-slam-detection/releases/tag/v1.3.0