w111liang222/lidar-slam-detection #21: can reltime show lidar data

yanyong666, 2024-01-25:
can reltime show lidar data

w111liang222, 2024-01-26:
We tested the LSD can visualize 20w+ points of 10Hz on a normal PC

zhu2900000, 2024-05-22:
你好，怎么才能接入自己的激光雷达？是否有具体的方法。在网页的配置页面就能实现？不需要配ip吗

w111liang222, 2024-05-22:
> 你好，怎么才能接入自己的激光雷达？是否有具体的方法。在网页的配置页面就能实现？不需要配ip吗

1. 可以接入已有驱动的雷达，设置雷达目标地址到你的设备上
2. 没有现在驱动的可以使用rosbag_proxy转发或者自己在sensor_driver/lidar_driver中集成对应的驱动

zhu2900000, 2024-05-23:
非常感谢。我看了一下我的lidar应该是内置驱动的，因为上面有目的地址的ip，我把本机配置了对应的ip但是收不到数据 能否帮我看看。
[image]
[image]
[image]
[image]

w111liang222, 2024-05-23:
> 非常感谢。我看了一下我的lidar应该是内置驱动的，因为上面有目的地址的ip，我把本机配置了对应的ip但是收不到数据 能否帮我看看。 ![image](https://private-user-images.githubusercontent.com/36877360/333005010-e0f6f0fb-d494-41a7-9350-330b791678ae.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY0Mzk1MjUsIm5iZiI6MTcxNjQzOTIyNSwicGF0aCI6Ii8zNjg3NzM2MC8zMzMwMDUwMTAtZTBmNmYwZmItZDQ5NC00MWE3LTkzNTAtMzMwYjc5MTY3OGFlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTIzVDA0NDAyNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWFmNjcxNzc0NDExYWVhNzcwMWZlN2Q5YWZkZTcwZTU1MjJhMGZlM2I0YTE1MzQ5Y2FjYjVhMmRhNTBkNDg3OGYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.dk2JxmYObIsY55ONwsrfJ0TSVbcDEqUiud0dUe6Aawc) ![image](https://private-user-images.githubusercontent.com/36877360/333005032-e81fde18-e320-4e97-8881-32ae01e865d9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY0Mzk1MjUsIm5iZiI6MTcxNjQzOTIyNSwicGF0aCI6Ii8zNjg3NzM2MC8zMzMwMDUwMzItZTgxZmRlMTgtZTMyMC00ZTk3LTg4ODEtMzJhZTAxZTg2NWQ5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTIzVDA0NDAyNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTJkMTA2ZGY5MTViYWViNWU3MzE2OTY3M2FmM2RiMTk3ZjQ0YmQ5YWEwOTdiYzQzNmJlZDI0NzExMDI4OWYyZmYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.RXFbLHPRUbAZhNYC1_HkPF2sdLSdMsmdkpYwEjbJYzY) ![image](https://private-user-images.githubusercontent.com/36877360/333005051-7355aed1-cb96-45df-9ad6-66d95759a856.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY0Mzk1MjUsIm5iZiI6MTcxNjQzOTIyNSwicGF0aCI6Ii8zNjg3NzM2MC8zMzMwMDUwNTEtNzM1NWFlZDEtY2I5Ni00NWRmLTlhZDYtNjZkOTU3NTlhODU2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTIzVDA0NDAyNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM1ZDAwMzc0OGFiM2IzYjMzMTEzZDUyODQ2YzcyZmI2YTFlZTEzMGM0Mjc3MTExYmQyOGU1ZDI2OWEzMGUyMTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.HIkfjkKWnGVwVyIzlZaTmGXIX133jmnENahpFxrUi_E) ![DINGTALK_IM_1555416159 JPG](https://private-user-images.githubusercontent.com/36877360/333006007-903ec166-5438-4c8a-b3f6-e64a0ad1eb83.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY0Mzk1MjUsIm5iZiI6MTcxNjQzOTIyNSwicGF0aCI6Ii8zNjg3NzM2MC8zMzMwMDYwMDctOTAzZWMxNjYtNTQzOC00YzhhLWIzZjYtZTY0YTBhZDFlYjgzLkpQRz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTIzVDA0NDAyNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWNmYmU3NmZmYmE1NzkwMTBjNDUyOTkwYWNmNTVkYmEwMzg2ODJhMGNjNGU1MWUxYjBlYzE5NGRhN2QzYmQ3YzkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.ymzqF4YgGuUdzJLOebjNth98tph-i3_2z9E7wN5bBb0)

tcpdump看看有没有数据，另外RS的激光DIFOP端口要设置成数据端口号+1

zhu2900000, 2024-05-24:
@w111liang222 麻烦把上面回复里第一张图删了可以吗？有mac地址的那个，谢谢了

zhu2900000, 2024-05-24:
[image]
[image]
你好，我用的cuda的docker，我的显卡是特斯拉p4*2,现在界面刷不出来，一直在初始化中，请问是什么问题？谢谢！

zhu2900000, 2024-05-25:
> > 你好，怎么才能接入自己的激光雷达？是否有具体的方法。在网页的配置页面就能实现？不需要配ip吗
> 
> 1. 可以接入已有驱动的雷达，设置雷达目标地址到你的设备上
> 2. 没有现在驱动的可以使用rosbag_proxy转发或者自己在sensor_driver/lidar_driver中集成对应的驱动

@w111liang222 你好，我看界面上有RS Helios_16P这个型号，应该算LSD已经支持的驱动吧？ 我手里的设备就是这款。如果收不到应该排查网络问题对吗？

w111liang222, 2024-05-25:
> ![image](https://private-user-images.githubusercontent.com/36877360/333521575-1ccdb66d-e2c4-4fa4-9c59-d5dc77816949.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

启动的时候应该有报错，终端log拉到上面看看

zhu2900000, 2024-05-27:
你好，以下是tcpdump -i ens192 -vnn src port 6699命令和启动程序的log，现在接不到lidar数据，能否帮忙分析下，谢谢！

01:26:28.870727 IP (tos 0x0, ttl 64, id 18839, offset 0, flags [DF], proto UDP (17), length 1276)
    10.11.109.247.6699 > 10.11.109.246.6699: UDP, length 1248
01:26:28.871689 IP (tos 0x0, ttl 64, id 18840, offset 0, flags [DF], proto UDP (17), length 1276)
    10.11.109.247.6699 > 10.11.109.246.6699: UDP, length 1248
01:26:28.871818 IP (tos 0x0, ttl 64, id 18841, offset 0, flags [DF], proto UDP (17), length 1276)
    10.11.109.247.6699 > 10.11.109.246.6699: UDP, length 1248
01:26:28.873748 IP (tos 0x0, ttl 64, id 18842, offset 0, flags [DF], proto UDP (17), length 1276)
    10.11.109.247.6699 > 10.11.109.246.6699: UDP, length 1248
01:26:28.873885 IP (tos 0x0, ttl 64, id 18843, offset 0, flags [DF], proto UDP (17), length 1276)
    10.11.109.247.6699 > 10.11.109.246.6699: UDP, length 1248
01:26:28.874212 IP (tos 0x0, ttl 64, id 18844, offset 0, flags [DF], proto UDP (17), length 1276)
    10.11.109.247.6699 > 10.11.109.246.6699: UDP, length 1248
01:26:28.875697 IP (tos 0x0, ttl 64, id 18845, offset 0, flags [DF], proto UDP (17), length 1276)
    10.11.109.247.6699 > 10.11.109.246.6699: UDP, length 1248
01:26:28.875809 IP (tos 0x0, ttl 64, id 18846, offset 0, flags [DF], proto UDP (17), length 1276)
    10.11.109.247.6699 > 10.11.109.246.6699: UDP, length 1248
01:26:28.875993 IP (tos 0x0, ttl 64, id 18847, offset 0, flags [DF], proto UDP (17), length 1276)
    10.11.109.247.6699 > 10.11.109.246.6699: UDP, length 1248
01:26:28.876877 IP (tos 0x0, ttl 64, id 18848, offset 0, flags [DF], proto UDP (17), length 1276)
    10.11.109.247.6699 > 10.11.109.246.6699: UDP, length 1248
01:26:28.876985 IP (tos 0x0, ttl 64, id 18849, offset 0, flags [DF], proto UDP (17), length 1276)
    10.11.109.247.6699 > 10.11.109.246.6699: UDP, length 1248
01:26:28.877876 IP (tos 0x0, ttl 64, id 18850, offset 0, flags [DF], proto UDP (17), length 1276)
    10.11.109.247.6699 > 10.11.109.246.6699: UDP, length 1248
01:26:28.878957 IP (tos 0x0, ttl 64, id 18851, offset 0, flags [DF], proto UDP (17), length 1276)
    10.11.109.247.6699 > 10.11.109.246.6699: UDP, length 1248
01:26:28.879209 IP (tos 0x0, ttl 64, id 18852, offset 0, flags [DF], proto UDP (17), length 1276)
    10.11.109.247.6699 > 10.11.109.246.6699: UDP, length 1248
01:26:28.880131 IP (tos 0x0, ttl 64, id 18853, offset 0, flags [DF], proto UDP (17), length 1276)
    10.11.109.247.6699 > 10.11.109.246.6699: UDP, length 1248
01:26:28.880312 IP (tos 0x0, ttl 64, id 18854, offset 0, flags [DF], proto UDP (17), length 1276)
^C    10.11.109.247.6699 > 10.11.109.246.6699: UDP, length 1248

run server python script
dpkg-query: no packages found matching nvidia-l4t-core
Running on board: v2.1.0-x86_64-x86_64-20.04
2024-05-27 01:21:28,027[3804:3804][module_manager.py: 47] pipeline: Source connect to Sink
2024-05-27 01:21:28,027[3804:3804][module_manager.py: 71] start to setup module: Sink
2024-05-27 01:21:28,039[3804:3804][module_manager.py: 73] setup module: Sink, done
2024-05-27 01:21:28,039[3804:3804][module_manager.py: 71] start to setup module: Source
2024-05-27 01:21:28,040[3804:3804][source_manager.py: 31] Source, start to setup: lidar
2024-05-27 01:21:28,041[3815:3815][sink_template.py: 50] deamon UdpSink starts
2024-05-27 01:21:28.046[3804:3804][lidar_driver_wrapper.cpp:34] lidar 0:RS-Helios open success
2024-05-27 01:21:28.046[3804:3804][lidar_driver_wrapper.cpp:46] set lidar 0:RS-Helios external parameter [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
2024-05-27 01:21:28.046[3804:3804][lidar_driver_wrapper.cpp:58] set lidar 0:RS-Helios range filter [-72.0, -72.0, -2.0, 72.0, 72.0, 4.0]
2024-05-27 01:21:28.046[3804:3804][lidar_driver_wrapper.cpp:70] set lidar 0:RS-Helios exclude [-0.5, -0.5, 0.0, 0.5, 0.5, 0.0]
2024-05-27 01:21:28,046[3804:3804][source_manager.py: 34] Source, setup: lidar, done
2024-05-27 01:21:28,046[3804:3804][source_manager.py: 31] Source, start to setup: camera
2024-05-27 01:21:28,107[3804:3804][camera_data_manager.py:195] Camera: 0, try cap: v4l2src device=/dev/video0 ! video/x-raw   ! videoscale ! video/x-raw,width=1920,height=1080  ! videoconvert ! video/x-raw,width=1920,height=1080,format=I420 ! tee name=t
    t. ! queue ! jpegenc ! shmsink socket-path=/tmp/camera_jpeg_0 sync=false
    t. ! queue ! appsink sync=false 
[ WARN:0@0.587] global /tmp/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (2401) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src0 reported: Cannot identify device '/dev/video0'.
[ WARN:0@0.587] global /tmp/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (1356) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0@0.587] global /tmp/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (862) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
2024-05-27 01:21:28,149[3804:3804][camera_data_manager.py:198] Camera: 0, can not open
2024-05-27 01:21:28,149[3804:3804][camera_data_manager.py:195] Camera: 1, try cap: v4l2src device=/dev/video1 ! video/x-raw   ! videoscale ! video/x-raw,width=1920,height=1080  ! videoconvert ! video/x-raw,width=1920,height=1080,format=I420 ! tee name=t
    t. ! queue ! jpegenc ! shmsink socket-path=/tmp/camera_jpeg_1 sync=false
    t. ! queue ! appsink sync=false 
[... cut ...]
2024-05-27 01:21:39,333[3804:3859][module_manager.py:137] Process 3815:UdpSink    CPU   0.0%, Mem 128 MB
2024-05-27 01:21:39,435[3804:3859][module_manager.py:137] Process 3804:Perception CPU   0.0%, Mem 222 MB
2024-05-27 01:21:40.214[3804:3857][lidar_driver_wrapper.cpp:138] 0:RS-Helios No lidar data comes
2024-05-27 01:21:40,215[3804:3857][data_manager_template.py: 62] Lidar: 0-RS-Helios, acquire FPS: 1.0
2024-05-27 01:21:40,218[3804:3856][manager_template.py: 84] Source has no data
2024-05-27 01:21:40,219[3804:3856][manager_template.py: 80] Source, FPS: 1.0

zhu2900000, 2024-05-27:
> > 你好，怎么才能接入自己的激光雷达？是否有具体的方法。在网页的配置页面就能实现？不需要配ip吗
> 
>     1. 可以接入已有驱动的雷达，设置雷达目标地址到你的设备上
> 
>     2. 没有现在驱动的可以使用rosbag_proxy转发或者自己在sensor_driver/lidar_driver中集成对应的驱动

你好，是端口的问题，我搞定了,非常感谢！  6699 6700哈
@w111liang222