w111liang222/lidar-slam-detection #22: How to solve the problem of not finding the model at startup ？

ohm-zx, 2024-02-22:
ModuleNotFoundError: No module named 'cpp_utils_ext'
Traceback (most recent call last):
File "web_backend/server.py", line 36, in &lt;module&gt;
from perception_server import PerceptionServer
File "/home/nvidia/znqc/work/lidar-slam-detection/web_backend/perception_server.py", line 5, in &lt;module&gt;
from jsonrpc.backend.flask import api
ModuleNotFoundError: No module named 'jsonrpc.backend'

w111liang222, 2024-02-22:
Hi
1. did you run the 'python setup.py install' before startup, the cpp_utils_ext is a cpp extension package in this project
2. 'pip install json-rpc' to install the jsonrpc module

ohm-zx, 2024-03-06:
Thanks，i also want to ask the following question
ModuleNotFoundError: No module named 'zerocm'

w111liang222, 2024-03-06:
> Thanks，i also want to ask the following question ModuleNotFoundError: No module named 'zerocm'

'./tools/scripts/install_dependency.sh' to install the python extension of zcm

ohm-zx, 2024-03-07:
Thanks，please help me look the following question
[Errno 13] Permission denied: '/home/znqc/work/licence/licences'
Traceback (most recent call last):
  File "web_backend/server.py", line 48, in &lt;module&gt;
    @app.before_first_request
AttributeError: 'Flask' object has no attribute 'before_first_request'

w111liang222, 2024-03-08:
> Thanks，please help me look the following question [Errno 13] Permission denied: '/home/znqc/work/licence/licences' Traceback (most recent call last): File "web_backend/server.py", line 48, in <module> @app.before_first_request AttributeError: 'Flask' object has no attribute 'before_first_request'

please use the 2.x version of flask