w111liang222/lidar-slam-detection #23: How can I convert the code in the project(instead of visualization and web modules) into C++?

scottymax, 2024-02-23:
Thanks to your great job! I want to combine this project with autoware planning core, and i found every main nodes was written by python. Do you have some ideas to convert them to C++? Thank you!

w111liang222, 2024-02-23:
Thank you for your attention to this project. 

The python code here is responsible only for data transmission and some basic logic. CPU-intensive core algorithms are implemented in C++. We have tested it on Orin NX/AGX and it can process 1 128-line lidar and 4 or more 1080P cameras at 10Hz. 

You can send the processing output of LSD via UDP, and create a receiving and transforming node in Autoware to  meet your need.

scottymax, 2024-02-28:
> Thank you for your attention to this project.
>
> The python code here is responsible only for data transmission and some basic logic. CPU-intensive core algorithms are implemented in C++. We have [...]

谢谢，我现在成功部署了CPU版本，但是部署gpu版本出现config进不去的问题，机器配置是3080，cuda11.4，docker运行也正常，请问哪里可以排查问题？另外我在测试demo的时候循环播放demo数据，之前的地图会保留在TVIZ上并和当前帧数据不对齐。。看着也有点奇怪。

w111liang222, 2024-02-29:
1. GPU版本的应该是程序启动时阻塞在import slam语句上了，之前的issue里有提到过，但是我在我这边的机器上一直没有复现这个问题，所以具体原因还不清楚
2. 循环播放bag会存在放完一轮后，从头回放的问题，在lsd里会检查时间戳是否是连续的，所以出现了你说的现象，不影响实际建图效果，你可以不循环放就可以了

scottymax, 2024-03-27:
> 1. GPU版本的应该是程序启动时阻塞在import slam语句上了，之前的issue里有提到过，但是我在我这边的机器上一直没有复现这个问题，所以具体原因还不清楚
> 
>     2. 循环播放bag会存在放完一轮后，从头回放的问题，在lsd里会检查时间戳是否是连续的，所以出现了你说的现象，不影响实际建图效果，你可以不循环放就可以了

非常感谢解答，目前我正在学习您的代码，python部分我还比较弱，有一些问题想要请教。module_manager默认的config是slam、sink在order_dict前面，最后放source，那传输数据到peer的过程是怎么由source传输到slam的？另外如果要同时开启slam的定位功能和detection功能，source里的数据只保存在一个队列，怎么传递给多个消费者？您在硬件需求中说显卡要SM80+，我看了一下只有3090和4090的流处理器单元满足需求，那是不是90系列以下的卡跑带GPU的版本都会出现config加载不出来的情况。。

w111liang222, 2024-03-27:
> 非常感谢解答，目前我正在学习您的代码，python部分我还比较弱，有一些问题想要请教。module_manager默认的config是slam、sink在order_dict前面，最后放source，那传输数据到peer的过程是怎么由source传输到slam的？另外如果要同时开启slam的定位功能和detection功能，source里的数据只保存在一个队列，怎么传递给多个消费者？您在硬件需求中 [...]

你好，
1. order_dict的存储顺序不是实际数据流的顺序，会有一个setup_pipeline的调用，根据配置文件中写的顺序，默认是Source -> SLAM -> Sink，如果是同时开启目标检测和建图定位，会使用两个额外的模块是Splitter和Merger(在module/common目录下)，负责把数据分开和合并
2. SM80+是目标检测模块的要求(因为用到了nvidia开源的libspconv.so)，SLAM模块没有这个要求，如果只使用SLAM功能，运行在其他的GPU上时，需要在几个CMakeLists.txt里的编译参数加上对应GPU的arch code

scottymax, 2024-03-28:
> 你好，
>
>     1. order_dict的存储顺序不是实际数据流的顺序，会有一个setup_pipeline的调用，根据配置文件中写的顺序，默认是Source -> SLAM -> Sink，如果是同时开启目标检测和建图定位，会使用两个额外的模块是Splitter和Merger(在module/common目录下)，负责把数据分开和合并
>
>     2. [...]

也就是说如果要同时开启定位和感知，我需要在cfg的Pipeline里加上Splitter和Merger是吗？请问这个顺序有没有要求？

scottymax, 2024-03-28:
> 你好，
>
>     1. order_dict的存储顺序不是实际数据流的顺序，会有一个setup_pipeline的调用，根据配置文件中写的顺序，默认是Source -> SLAM -> Sink，如果是同时开启目标检测和建图定位，会使用两个额外的模块是Splitter和Merger(在module/common目录下)，负责把数据分开和合并
>
>     2. [...]

[image]
[image]
目前我这么配置cfg，出来的pipeline是source-slam-detect-sink。。

w111liang222, 2024-03-28:
> > > > 1. GPU版本的应该是程序启动时阻塞在import slam语句上了，之前的issue里有提到过，但是我在我这边的机器上一直没有复现这个问题，所以具体原因还不清楚
> > > >    
> > > >    2. 循环播放bag会存在放完一轮后，从头回放的问题，在lsd里会检查时间戳是否是连续的，所以出现了你说的现象，不影响实际建图效果，你可以不循环放就可以了
> > > 
> > > 
> > > 非常感谢解答，目前我正在学习您的代码，python部分我还比较弱，有一些问题想要请教。module_manager默认的config是slam、sink在order_dict前面，最后放source，那传输数据到peer的过程是怎么由source传输到slam的？另外如果要同时开启slam的定位功能和detection功能，source里的数据只保存在一个队列，怎么传递给多个消费者？您在硬件需求中说显卡要SM80+，我看了一下只有3090和4090的流处理器单元满足需求，那是不是90系列以下的卡跑带GPU的版本都会出现config加载不出来的情况。。
> > 
> > 
> > 你好，
> > ```
> > 1. order_dict的存储顺序不是实际数据流的顺序，会有一个setup_pipeline的调用，根据配置文件中写的顺序，默认是Source -> SLAM -> Sink，如果是同时开启目标检测和建图定位，会使用两个额外的模块是Splitter和Merger(在module/common目录下)，负责把数据分开和合并
> > 
> > 2. SM80+是目标检测模块的要求(因为用到了nvidia开源的libspconv.so)，SLAM模块没有这个要求，如果只使用SLAM功能，运行在其他的GPU上时，需要在几个CMakeLists.txt里的编译参数加上对应GPU的arch code
> > ```
> 
> ![图片](https://private-user-images.githubusercontent.com/44422324/317569811-02e62d6b-f6be-47df-9164-f026d7284824.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTE2MDI3OTksIm5iZiI6MTcxMTYwMjQ5OSwicGF0aCI6Ii80NDQyMjMyNC8zMTc1Njk4MTEtMDJlNjJkNmItZjZiZS00N2RmLTkxNjQtZjAyNmQ3Mjg0ODI0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzI4VDA1MDgxOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRkMDRhYTU2OGNmM2QzMTI5Nzc2YWJkZGI0YzUxYTBjOTNiZDI2MWJiOTcwNjc5NzlhYzE5ODJhYTUxYjUzZDMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.SAOahKNk0SihX2fop8JgaqbS4zQPaizhqtXSKzokIww) ![图片](https://private-user-images.githubusercontent.com/44422324/317569865-93dabf8a-26dc-4852-a894-497894227f5d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTE2MDI3OTksIm5iZiI6MTcxMTYwMjQ5OSwicGF0aCI6Ii80NDQyMjMyNC8zMTc1Njk4NjUtOTNkYWJmOGEtMjZkYy00ODUyLWE4OTQtNDk3ODk0MjI3ZjVkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzI4VDA1MDgxOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTIwN2ZmNDdkN2JjYjM1YTE1NTYwZTU5MTFlZmE2NDJkZTExYmM3MzliZDg3Njc4MjY3MzVkMmRhZGJmNDFkNTYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.8aS7_6DrYM7Z59kwo01bAcczEvvJJN8tE-VsuprESg0) 目前我这么配置cfg，出来的pipeline是source-slam-detect-sink。。

你好:

检测+建图模式的pipeline设置是下图这样的：
[image]

配置该模式时，可以通过Web页面来配置，参考https://github.com/w111liang222/lidar-slam-detection/blob/main/docs/guide.md，确认后需要重新运行下LSD

scottymax, 2024-04-16:
> 你好:
>
> 检测+建图模式的pipeline设置是下图这样的： [...]

谢谢大佬，docker版本这边没有什么问题了，现在我自己搭了整套lsd的环境，运行demo的时候遇到报错GStreamer warning: Error opening bin: no element "shmsink"，但是感知和建图跑起来没有问题，后面安装gstreamer的插件包后可以调用shmsink socket了，但是系统阻塞在open camera 2，请问有什么办法可以解决？我看warning里报的错好像是抓取视频流有问题...
[image]