w111liang222/lidar-slam-detection #25: can not run the demo

lucianzhong, 2024-04-24:
Hi:
Thanks for open source this great work. But I can not the demo. Here is a screen shot of my localhost in firefox.
Please have a look.
[image]

w111liang222, 2024-04-24:
Hi, 

please try use the non-cuda docker image to check if the problem caused by GPU

lucianzhong, 2024-04-24:
Thanks for the quick replay. I will try

jcpcxry, 2024-05-21:
> Hi,
> 
> please try use the non-cuda docker image to check if the problem caused by GPU

Hi,

I encountered the same issue. It works fine on the non-cuda Docker image, but there is no object detection. On the GPU version, I only get the following information:

root@slam:/home/znqc/work/lidar-slam-detection# tools/scripts/start_system.sh
run server python script
dpkg-query: no packages found matching nvidia-l4t-core
Running on board: v2.1.0-x86_64-x86_64-20.04

w111liang222, 2024-05-21:
> Hi,
>
> I encountered the same issue. It works fine on the non-cuda Docker image, but there is no object detection. On the GPU version, I only get the following information: [...]

1. you can remove the code of 'import slam' related code if you just want to test the object detection.
2. check your GPU card，add the arch code into the CMakeLists to support your hardware