w111liang222/lidar-slam-detection #29: 单目3D算法在哪里

TYilin, 2024-05-14:
我已经成功运行demo，但我把demo中detection里面lidar输入删除后，就没有detected objects,然后看了一下detection的代码，sensor_inference/object_infer.py,在这没看到单目3D的算法，只看到了点云检测的部分，Fusion部分我看介绍中谈到的是单目3D与点云检测的融合，但我看了sensor_fusion/fusion.py，感觉是Trafficlight和object Detection的融合，具体是在哪里做的单目3D算法，能告知一下吗，没有太看懂

w111liang222, 2024-05-14:
在master分支上，单目3D已经删除了，单目3D请参考tag v1.3版本

lonlonago, 2024-10-22:
> D

那么请问现在是没有视觉的融合了吗？

w111liang222, 2024-10-22:
> > D
> 
> 那么请问现在是没有视觉的融合了吗？

main分支已经移除单目3D了

lonlonago, 2024-10-22:
> > 那么请问现在是没有视觉的融合了吗？
> 
> main分支已经移除单目3D了

谢谢您的回复，我还想问问：
移除的原因是什么呢？
1.3的版本是可以使用CPU的版本的雷达检测和单目3D融合吗？  
是否可以不编译和使用slam的模块呢？   
打扰了，谢谢回复~

w111liang222, 2024-10-22:
> > > 那么请问现在是没有视觉的融合了吗？
> > 
> > 
> > main分支已经移除单目3D了
> 
> 谢谢您的回复，我还想问问： 移除的原因是什么呢？ 1.3的版本是可以使用CPU的版本的雷达检测和单目3D融合吗？ 是否可以不编译和使用slam的模块呢？ 打扰了，谢谢回复~

1. 单目3D检测精度不高，本来是打算替换成多相机BEV检测模型的
2. slam算法模块是c++实现的，需要编译

lonlonago, 2024-10-22:
> > > > 那么请问现在是没有视觉的融合了吗？
> > > 
> > > 
> > > main分支已经移除单目3D了
> > 
> > 
> > 谢谢您的回复，我还想问问： 移除的原因是什么呢？ 1.3的版本是可以使用CPU的版本的雷达检测和单目3D融合吗？ 是否可以不编译和使用slam的模块呢？ 打扰了，谢谢回复~
> 
> 1. 单目3D检测精度不高，本来是打算替换成多相机BEV检测模型的
> 2. slam算法模块是c++实现的，需要编译

懂了，谢谢了，是准备做BEV还没做完是吧

我上面第三个问题再准确表述一下，我不是问是否需要编译的意思，是说您的这个项目是否可以整体移除SLAM模块，我是想着不用SLAM，单独用您的1.3的激光雷达感知和相机感知融合的部分功能，然后想在CPU运行，不知道是否可行？

w111liang222, 2024-10-22:
> 懂了，谢谢了，是准备做BEV还没做完是吧
>
> 我上面第三个问题再准确表述一下，我不是问是否需要编译的意思，是说您的这个项目是否可以整体移除SLAM模块，我是想着不用SLAM，单独用您的1.3的激光雷达感知和相机感知融合的部分功能，然后想在CPU运行，不知道是否可行？

可以通过切换模式来选择只跑检测部分，但是感知检测是通过CNN模型输出的，纯CPU跑帧率会非常低

lonlonago, 2024-10-22:
> 可以通过切换模式来选择只跑检测部分，但是感知检测是通过CNN模型输出的，纯CPU跑帧率会非常低

慢一点我倒是可以接受，先试试，那是否可以不安装cuda来编译呢，主要是我的电脑安装这个不方便

谢谢回复~

w111liang222, 2024-10-22:
> 慢一点我倒是可以接受，先试试，那是否可以不安装cuda来编译呢，主要是我的电脑安装这个不方便
>
> 谢谢回复~

you can use the cuda-free version docker image: **docker pull 15liangwang/auto-ipu**

lonlonago, 2024-10-22:
> you can use the cuda-free version docker image: **docker pull 15liangwang/auto-ipu**

对了，最后想问下，这个镜像的代码和version1.3的是不一样的吗？  还是说我要1.3的自己编译也是一样的

w111liang222, 2024-10-22:
> 对了，最后想问下，这个镜像的代码和version1.3的是不一样的吗？ 还是说我要1.3的自己编译也是一样的

cuda-free的镜像是通用的，1.3和main的代码都可以编译运行

lonlonago, 2024-10-22:
> 1.3

我的意思是cuda-free的镜像就是1.3的代码编译出来的吗？ 因为我看1.3的说明里面还是用到了tensorrt的，好像还不是cuda-free的