w111liang222/lidar-slam-detection #31: 无法调用显卡加速1.3.1

zhu2900000, 2024-05-27:
你好，我用的1.3.1版本，切换到目标检测模式后，可输出结果，但是好像没有用到显卡，检测帧率很低（速度慢），请问应该怎么调试以下，我的docker用的gpu版本

[image]

un server python script
dpkg-query: no packages found matching nvidia-l4t-core
Running on board: IPC, machine: x86_64, jetpack: 20.04
2024-05-27 08:39:01,478[26853:26853][       infer.py: 23] image object detection is deactived
2024-05-27 08:39:01,490[26853:26853][module_manager.py: 47] pipeline: Source connect to Detect
2024-05-27 08:39:01,491[26853:26853][module_manager.py: 47] pipeline: Detect connect to Sink
2024-05-27 08:39:01,491[26853:26853][module_manager.py: 71] start to setup module: Detect
2024-05-27 08:39:01,506[26853:26853][object_filter.py: 25] class mapping: {'vehicle': 1, 'pedestrian': 2, 'cyclist': 3, 'traffic_cone': 4}
2024-05-27 08:39:01,506[26853:26853][module_manager.py: 73] setup module: Detect, done
2024-05-27 08:39:01,507[26853:26853][module_manager.py: 71] start to setup module: Sink
2024-05-27 08:39:01,513[26853:26853][module_manager.py: 73] setup module: Sink, done
2024-05-27 08:39:01,514[26853:26853][module_manager.py: 71] start to setup module: Source
2024-05-27 08:39:01,514[26874:26874][sink_template.py: 58] deamon UdpSink starts
2024-05-27 08:39:01,523[26853:26853][player_data_manager.py:135] player have 201 frames, duration 19.971323 s
/usr/local/lib/python3.8/dist-packages/onnxruntime/capi/onnxruntime_inference_collection.py:65: UserWarning: Specified provider 'OpenVINOExecutionProvider' is not in available provider names.Available providers: 'CPUExecutionProvider'
  warnings.warn(
 * Serving Flask app 'module.source.player_data_manager'
 * Debug mode: off
2024-05-27 08:39:01,724[26853:26853][source_manager.py: 35] Source, start to setup: lidar
WARNING: This is a development server. Do not use it in a production deployment. Use a production WSGI server instead.
 * Running on http://127.0.0.1:38000
Press CTRL+C to quit
2024-05-27 08:39:01.738[26853:26853][lidar_driver_wrapper.cpp:34] lidar 0:Ouster-OS1-64 open success
2024-05-27 08:39:01.738[26853:26853][lidar_driver_wrapper.cpp:46] set lidar 0:Ouster-OS1-64 external parameter [-0.03368380293250084, -0.660768985748291, 1.7658969163894653, 177.67449951171875, 135.47140502929688, -177.0221710205078]
2024-05-27 08:39:01.738[26853:26853][lidar_driver_wrapper.cpp:58] set lidar 0:Ouster-OS1-64 range filter [-72.0, -72.0, -2.0, 72.0, 72.0, 4.0]
2024-05-27 08:39:01.738[26853:26853][lidar_driver_wrapper.cpp:70] set lidar 0:Ouster-OS1-64 exclude [-2.13, -2.01, -5.0, 2.92, 0.5, 5.0]
2024-05-27 08:39:01.745[26853:26853][lidar_driver_wrapper.cpp:34] lidar 1:Ouster-OS1-128 open success
2024-05-27 08:39:01.745[26853:26853][lidar_driver_wrapper.cpp:46] set lidar 1:Ouster-OS1-128 external parameter [0.0, 0.0, 1.88961923122406, 179.99972534179688, 179.9855194091797, -177.86497497558594]
2024-05-27 08:39:01.745[26853:26853][lidar_driver_wrapper.cpp:58] set lidar 1:Ouster-OS1-128 range filter [-72.0, -72.0, -2.0, 72.0, 72.0, 4.0]
2024-05-27 08:39:01.745[26853:26853][lidar_driver_wrapper.cpp:70] set lidar 1:Ouster-OS1-128 exclude [-3.37, -3.02, -4.15, 3.16, 2.33, 4.7]
2024-05-27 08:39:01.751[26853:26853][lidar_driver_wrapper.cpp:34] lidar 2:Ouster-OS1-64 open success
2024-05-27 08:39:01.751[26853:26853][lidar_driver_wrapper.cpp:46] set lidar 2:Ouster-OS1-64 external parameter [-0.09995567798614502, 0.6768492460250854, 1.7785784006118774, 1.0974562168121338, -44.20530319213867, 2.35439395904541]
2024-05-27 08:39:01.751[26853:26853][lidar_driver_wrapper.cpp:58] set lidar 2:Ouster-OS1-64 range filter [-72.0, -72.0, -2.0, 72.0, 72.0, 4.0]
2024-05-27 08:39:01.751[26853:26853][lidar_driver_wrapper.cpp:70] set lidar 2:Ouster-OS1-64 exclude [-3.37, -3.14, -4.92, 3.16, 1.44, 4.58]
2024-05-27 08:39:01,751[26853:26853][source_manager.py: 38] Source, setup: lidar, done
2024-05-27 08:39:01,751[26853:26853][source_manager.py: 35] Source, start to setup: camera
2024-05-27 08:39:01,807[26853:26853][camera_data_manager.py:207] camera 0, try cap: souphttpsrc timeout=0 location=http://127.0.0.1:38000/v1/camera/0 ! jpegparse ! jpegdec ! video/x-raw  ! videoscale ! video/x-raw,width=1920,height=1080  ! videoconvert ! video/x-raw,width=1920,height=1080,format=I420 ! tee name=t
    t. ! queue ! jpegenc ! shmsink socket-path=/tmp/camera_jpeg_0 sync=false
    t. ! queue ! appsink sync=false 
127.0.0.1 - - [27/May/2024 08:39:01] "GET /v1/camera/0 HTTP/1.1" 200 -
[ WARN:0@0.925] global /tmp/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (1405) open OpenCV | GStreamer warning: Cannot query video position: status=1, value=0, duration=-1
2024-05-27 08:39:01,911[26853:26853][camera_data_manager.py:217] camera 0 open success
2024-05-27 08:39:01,911[26853:26853][camera_data_manager.py:207] camera 1, try cap: souphttpsrc timeout=0 location=http://127.0.0.1:38000/v1/camera/1 ! jpegparse ! jpegdec ! video/x-raw  ! videoscale ! video/x-raw,width=1920,height=1080  ! videoconvert ! video/x-raw,width=1920,height=1080,format=I420 ! tee name=t
    t. ! queue ! jpegenc ! shmsink socket-path=/tmp/camera_jpeg_1 sync=false
    t. ! queue ! appsink sync=false 
127.0.0.1 - - [27/May/2024 08:39:01] "GET /v1/camera/1 HTTP/1.1" 200 -
[ WARN:0@0.979] global /tmp/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (1405) open OpenCV | GStreamer warning: Cannot query video position: status=1, value=0, duration=-1
2024-05-27 08:39:01,965[26853:26853][camera_data_manager.py:217] camera 1 open success
2024-05-27 08:39:01,965[26853:26853][camera_data_manager.py:207] camera 2, try cap: souphttpsrc timeout=0 location=http://127.0.0.1:38000/v1/camera/2 ! jpegparse ! jpegdec ! video/x-raw  ! videoscale ! video/x-raw,width=1920,height=1080  ! videoconvert ! video/x-raw,width=1920,height=1080,format=I420 ! tee name=t
    t. ! queue ! jpegenc ! shmsink socket-path=/tmp/camera_jpeg_2 sync=false
    t. ! queue ! appsink sync=false 
127.0.0.1 - - [27/May/2024 08:39:04] "GET /v1/camera/2 HTTP/1.1" 200 -
[ WARN:0@3.389] global /tmp/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (1405) open OpenCV | GStreamer warning: Cannot query video position: status=1, value=0, duration=-1
2024-05-27 08:39:04,374[26853:26853][camera_data_manager.py:217] camera 2 open success
2024-05-27 08:39:04,374[26853:26853][camera_data_manager.py:207] camera 3, try cap: souphttpsrc timeout=0 location=http://127.0.0.1:38000/v1/camera/3 ! jpegparse ! jpegdec ! video/x-raw  ! videoscale ! video/x-raw,width=1920,height=1080  ! videoconvert ! video/x-raw,width=1920,height=1080,format=I420 ! tee name=t
    t. ! queue ! jpegenc ! shmsink socket-path=/tmp/camera_jpeg_3 sync=false
    t. ! queue ! appsink sync=false 
127.0.0.1 - - [27/May/2024 08:39:04] "GET /v1/camera/3 HTTP/1.1" 200 -
[ WARN:0@3.442] global /tmp/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (1405) open OpenCV | GStreamer warning: Cannot query video position: status=1, value=0, duration=-1
2024-05-27 08:39:04,427[26853:26853][camera_data_manager.py:217] camera 3 open success
2024-05-27 08:39:04,428[26853:26853][source_manager.py: 38] Source, setup: camera, done
2024-05-27 08:39:04,428[26853:26853][source_manager.py: 35] Source, start to setup: radar
2024-05-27 08:39:04,428[26853:26853][source_manager.py: 38] Source, setup: radar, done
2024-05-27 08:39:04,428[26853:26853][source_manager.py: 35] Source, start to setup: ins
2024-05-27 08:39:04.428[26853:26853][ins_driver_wrapper.cpp:19] ins open success
2024-05-27 08:39:04.428[26853:26853][ins_driver_wrapper.cpp:34] set ins external parameter [0.09695542603731155, 1.2899638414382935, -0.11553408205509186, 87.35172271728516, -0.02054435946047306, -0.8822724223136902]
[... 221 lines cut ...]
  File "src/gevent/_gevent_c_greenlet_primitives.pxd", line 35, in gevent._gevent_c_greenlet_primitives._greenlet_switch
KeyboardInterrupt
2024-05-27 08:39:13,777[26853:26955][manager_template.py: 80] Source, FPS: 1.5
2024-05-27 08:39:13,781[26853:26959][camera_data_manager.py:228] camera 2, get invalid data
2024-05-27 08:39:13,781[26853:26960][camera_data_manager.py:228] camera 3, get invalid data
2024-05-27 08:39:13,781[26853:26958][camera_data_manager.py:228] camera 1, get invalid data
2024-05-27 08:39:13,781[26853:26956][camera_data_manager.py:228] camera 0, get invalid data
Process Process-1:
Traceback (most recent call last):
  File "/usr/lib/python3.8/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/usr/lib/python3.8/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
[... cut ...]
  File "src/gevent/_greenlet_primitives.py", line 61, in gevent._gevent_c_greenlet_primitives.SwitchOutGreenletWithLoop.switch
  File "src/gevent/_greenlet_primitives.py", line 61, in gevent._gevent_c_greenlet_primitives.SwitchOutGreenletWithLoop.switch
  File "src/gevent/_greenlet_primitives.py", line 65, in gevent._gevent_c_greenlet_primitives.SwitchOutGreenletWithLoop.switch
  File "src/gevent/_gevent_c_greenlet_primitives.pxd", line 35, in gevent._gevent_c_greenlet_primitives._greenlet_switch

w111liang222, 2024-05-27:
v1.3.1版本目标检测在x86上目前只能运行在CPU上(基于openvino)，需要自行将onnx文件转换成tensorrt来调用显卡加速

zhu2900000, 2024-05-27:
> v1.3.1版本目标检测在x86上目前只能运行在CPU上(基于openvino)，需要自行将onnx文件转换成tensorrt来调用显卡加速
谢谢，master分支的支持吗？

zhu2900000, 2024-05-27:
> v1.3.1版本目标检测在x86上目前只能运行在CPU上(基于openvino)，需要自行将onnx文件转换成tensorrt来调用显卡加速
谢谢，master分支的支持吗？
@w111liang222

w111liang222, 2024-05-27:
master分支是支持的，但对显卡有要求，需要SM80以上架构的，你的tesla P4好像不满足。。

zhu2900000, 2024-05-28:
> master分支是支持的，但对显卡有要求，需要SM80以上架构的，你的tesla P4好像不满足。。

谢谢，如果我想让master分支支持p4显卡，应该如何修改代码，有建议吗？