w111liang222/lidar-slam-detection #32: 3d单目视觉调用

zhu2900000, 2024-05-28:
你好,我在调用3D单目视觉检测算法,我发现1.3.1版本没有打开,当我打开以后引起加载报错,请帮忙看一下.我想在GPU上跑双传感器融合的版本.我的GPU是特斯拉P4,
[image]
[image]

zhu2900000, 2024-05-28:
[image]

zhu2900000, 2024-05-28:
@w111liang222 你好,我更换方法,使用onnxruntime,在cpu上运行,加载模型 detection_image.onnx,  在计算目标框的时候回报错,原因是数据的尺度不对,我检查过模型的输入和实际输入的数据是一致的,请帮忙看一下:
[image]
这里是模型的输出,应该是KPS这个np的维度问题
[image]

模型的输入要求
[image]

视频输入的shape
[image]

zhu2900000, 2024-05-28:
我用的视频是demo的视频,加载的模型是sensor_inference/image_infer.py  版本是1.3.0

w111liang222, 2024-05-28:
参考https://github.com/w111liang222/lidar-slam-detection/blob/release_v1.3.0/sensor_inference/image_infer.py中的build_image_engine函数，将onnx文件生成tensorrt引擎，从而跑在gpu上

zhu2900000, 2024-05-28:
你好，我这里看不到
[image]

w111liang222, 2024-05-28:
> @w111liang222 你好,我更换方法,使用onnxruntime,在cpu上运行,加载模型 detection_image.onnx, 在计算目标框的时候回报错,原因是数据的尺度不对,我检查过模型的输入和实际输入的数据是一致的,请帮忙看一下: ![Screenshot from 2024-05-28 17-35-21](https://private-user-images.githubusercontent.com/36877360/334354168-9675b97a-22f6-41d8-a7ac-59928d75b3ed.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY5MDI0NDUsIm5iZiI6MTcxNjkwMjE0NSwicGF0aCI6Ii8zNjg3NzM2MC8zMzQzNTQxNjgtOTY3NWI5N2EtMjJmNi00MWQ4LWE3YWMtNTk5MjhkNzViM2VkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTI4VDEzMTU0NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM2MDY4YzQ1YTg1OWFhZjBkOTRjYmNlNGVlYWFhM2M2YmU1MDBhNWU0YjBmNTIxMjE2NTlkMTc1Yjc0ZjI2ZTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.p9npdMp01c7Ma_j1rNYvYFjAceQgFfqVtOKh6SQE3NQ) 这里是模型的输出,应该是KPS这个np的维度问题 ![Screenshot from 2024-05-28 17-36-40](https://private-user-images.githubusercontent.com/36877360/334354173-d7cc826c-c716-41e6-8ec4-7a16fd2fffb4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY5MDI0NDUsIm5iZiI6MTcxNjkwMjE0NSwicGF0aCI6Ii8zNjg3NzM2MC8zMzQzNTQxNzMtZDdjYzgyNmMtYzcxNi00MWU2LThlYzQtN2ExNmZkMmZmZmI0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTI4VDEzMTU0NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZhZmVjOTg1MzhhOTZjMDNmM2NkMDI2YmVmNjIzNTc0NWM0NWYyMWM4MTI5ZGI2ZDI5M2ZiNDQ3Nzc1MTE0M2EmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.stnZ3fihSAUL3EyHXLwWkm6f7CnvffzD00cZlTPJBxU)
> 
> 模型的输入要求 ![Screenshot from 2024-05-28 17-38-38](https://private-user-images.githubusercontent.com/36877360/334354277-382f2205-db41-4cec-94ea-e40a8f4f6f70.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY5MDI0NDUsIm5iZiI6MTcxNjkwMjE0NSwicGF0aCI6Ii8zNjg3NzM2MC8zMzQzNTQyNzctMzgyZjIyMDUtZGI0MS00Y2VjLTk0ZWEtZTQwYThmNGY2ZjcwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTI4VDEzMTU0NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTY0MzVhZTI1NTAzOGQ1ZTM2OTE3NGY4NzU1N2NlZTdiOTY2YzJjYWVkZTRkNWUyNTBkNDJkYmJmM2UzODI4YTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.VlVYSTfKEcxBkpi36T0wYZs9Bc3qERy3hJBlmXXagcE)
> 
> 视频输入的shape ![Screenshot from 2024-05-28 17-39-33](https://private-user-images.githubusercontent.com/36877360/334354172-99fbf0a0-f122-4221-a2c6-abe76fbaf8a9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY5MDI0NDUsIm5iZiI6MTcxNjkwMjE0NSwicGF0aCI6Ii8zNjg3NzM2MC8zMzQzNTQxNzItOTlmYmYwYTAtZjEyMi00MjIxLWEyYzYtYWJlNzZmYmFmOGE5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTI4VDEzMTU0NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTczMzc0MjYxZDJlNzVjZGY2NTcxOTVhNGJmNDk4YzFmYWRhNjUzMDlkMmFkZWI4NjZiNzI5NGI5MWNmYjgzZTkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.7XISniEvhn9w7UJhvqRmozgSKGvxRRMMNDQ1ebuAYDo)

kps的维度是1024x18，你可以通过https://netron.app网址加载image onnx查看输入输出，报错的原因估计是你的输出顺序搞错了

zhu2900000, 2024-05-29:
牛逼，我这里的输出顺序搞错了，和模型的输出对上就可以了，谢谢了！现在相机可以融合上了。 
[image]
另请问，咱们界面上可以单独看camera的结果吗？是否有单独调试的方法。
我看有一个传感器类型的选项，但是不知道如何区分。

zhu2900000, 2024-05-29:
> 1. 激光相机目标融合，依赖相机内外参，需要提前标定好(注意图像检测的输入尺寸是352x640，原始图像缩放对fx, fy, cx, cy的影响
> 
>     2. 按传感器类型显示后，不同颜色表示由不同传感器输出
>        3.可以打印proto/proto_serialize.py中的pred_object.sensor字段来确定由哪个传感器输出

你好，有几个疑问：
1、怎么才能看单独的相机的检测结果，在界面上看。
2、通过传感器类型显示，显示的是：融合的，lidar（单传感器未融合），camera（单传感器未融合）这样区分的吗？具体那个颜色和哪种类型对应？
3、我打印的pred_object.sensor只有0,我的lidar和camera都有数据，能正常融合。

zhu2900000, 2024-05-29:
> 1. 激光相机目标融合，依赖相机内外参，需要提前标定好(注意图像检测的输入尺寸是352x640，原始图像缩放对fx, fy, cx, cy的影响
> 
>     2. 按传感器类型显示后，不同颜色表示由不同传感器输出
>     3.可以打印proto/proto_serialize.py中的pred_object.sensor字段来确定由哪个传感器输出

@w111liang222 您好，关于第一个问题，您预测的很准。
[image]
3D的lidar 转成2D时候里面出现了空数据，我打印了内参发现里面是初始状态。我觉得问题出在这里。我跑的demo里自代的数据，cam的内参我没法标出来，我觉得您应该提供demo里数据对应的相机内参。

w111liang222, 2024-05-29:
不好意思。这个demo数据当时是只用来测试激光检测和激光slam的，由于时间太长了，现在已经找不到标定参数了。

zhu2900000, 2024-05-29:
@w111liang222 嗯嗯，很感谢，这么久还能提供支持，也是不容易啊。如果您之前跑过没有问题的话我就试一试，主要是怕里面有坑，和哈哈哈哈。

w111liang222, 2024-05-29:
功能肯定是没问题的，但是性能能不能达到你的预期就不一定了

zhu2900000, 2024-06-04:
> 1. 激光相机目标融合，依赖相机内外参，需要提前标定好(注意图像检测的输入尺寸是352x640，原始图像缩放对fx, fy, cx, cy的影响
你的意思是模型是在352x640上运行的，结果是基于352x640产生的，而iou匹配是基于原始图像投影的，需要把结果还原到原始图像下的坐标？我们的代码没实现吗？因为，视觉输入后都应该有尺寸变化的

zhu2900000, 2024-06-04:
我觉得咱们传感器数据进来以后应该统一和车辆后轴对齐，这样后面操作就好理解。否则每次操作还需要转来转去容易出错