w111liang222/lidar-slam-detection #33: 你好，相机标定模块问题

zhu2900000, 2024-05-30:
你好，我分别在目标检测模式下和数据采集模式下 通过rtsp接入相机进行标定，空格键保存大于10张，但是交点检测没有反应，状态一栏一直是空，一直无法到下一步，请帮忙看一下

zhu2900000, 2024-05-30:
> 你好，我分别在目标检测模式下和数据采集模式下 通过rtsp接入相机进行标定，空格键保存大于10张，但是交点检测没有反应，状态一栏一直是空，一直无法到下一步，请帮忙看一下

[image]

w111liang222, 2024-05-30:
棋盘格的长宽的格子数设置正确了吗？还有棋盘格的实际尺寸也需要设置，另外就是相机配置中的相机图像的宽高需要设置好

zhu2900000, 2024-05-31:
> 棋盘格的长宽的格子数设置正确了吗？还有棋盘格的实际尺寸也需要设置，另外就是相机配置中的相机图像的宽高需要设置好

以上按照您说的设置，不管用，另外我发现我的相机是265的编码，我修改了264,还是不行。

zhu2900000, 2024-06-03:
> 棋盘格的长宽的格子数设置正确了吗？还有棋盘格的实际尺寸也需要设置，另外就是相机配置中的相机图像的宽高需要设置好

老哥，你好，我标定的尺寸填的不对，我的格子是7*10,但是要填6*9的才对。虽然不知道为什么，麻烦你了。

w111liang222, 2024-06-03:
> > 棋盘格的长宽的格子数设置正确了吗？还有棋盘格的实际尺寸也需要设置，另外就是相机配置中的相机图像的宽高需要设置好
> 
> 老哥，你好，我标定的尺寸填的不对，我的格子是7_10,但是要填6_9的才对。虽然不知道为什么，麻烦你了。

这是opencv定义的，指的是棋盘格内部一圈的顶点个数