w111liang222/lidar-slam-detection #35: 关于修改ui

zhu2900000, 2024-06-17:
请问调整obj的显示ui如何才能生效？

w111liang222, 2024-06-17:
请参考web_ui下的README，npm run build后，将生成的dist文件夹把www替换，浏览器使用隐私模式访问，防止缓存不更新

zhu2900000, 2024-06-17:
/home/znqc/work/new/1.3.1geit/lidar-slam-detection/web_ui# npm run build

> webui-vite-ts-react@0.0.0 build /home/znqc/work/new/1.3.1geit/lidar-slam-detection/web_ui
> tsc -p tsconfig.production.json && VITE_COMMIT_HASH=`git rev-parse --short HEAD` vite build

internal/modules/cjs/loader.js:638
    throw err;
    ^

Error: Cannot find module 'worker_threads'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:636:15)
    at Function.Module._load (internal/modules/cjs/loader.js:562:25)
    at Module.require (internal/modules/cjs/loader.js:692:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at Object.<anonymous> (/home/znqc/work/new/1.3.1geit/lidar-slam-detection/web_ui/node_modules/vite/dist/node/chunks/dep-0a035c79.js:25:18)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! webui-vite-ts-react@0.0.0 build: `tsc -p tsconfig.production.json && VITE_COMMIT_HASH=`git rev-parse --short HEAD` vite build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the webui-vite-ts-react@0.0.0 build script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2024-06-17T05_15_06_701Z-debug.log

@w111liang222 hi 我尝试了该方法，没有生成dist 麻烦帮忙看一下