w111liang222/lidar-slam-detection #36: Cuda failure: CUDA driver version is insufficient for CUDA runtime version

Fernweh-yang, 2024-06-17:
Hi,  I have installed the latest NVIDIA Container Toolkit, but when I run the command "bash sensor_inference/pytorch_model/export/generate_trt.sh", I meet the problem:
[image]
Is there any advice? Thanks!

Fernweh-yang, 2024-06-17:
> 看起来像是cuda驱动版本太低了

我本地的cuda版本是12.2。不知道是不是镜像里的显卡驱动有问题？如果方便的话，能提供下dockerfile不?

w111liang222, 2024-06-17:
没有dockerfile，镜像是通过docker commit生成的，只有这个参考镜像

Fernweh-yang, 2024-06-17:
> 没有dockerfile，镜像是通过docker commit生成的，只有这个参考镜像

那这个镜像的主要功能是提供Ubuntu20.04, Python3.8, Eigen 3.3.7, Ceres 1.14.0, Protobuf 3.8.0, NLOPT 2.4.2, G2O, OpenCV 4.5.5, PCL 1.9.1, GTSAM 4.0吗？

MA-xiaowen, 2024-07-25:
> 没有dockerfile，镜像是通过docker commit生成的，只有这个参考镜像

请问这个出错有什么解决方法吗？

MA-xiaowen, 2024-07-25:
> > 看起来像是cuda驱动版本太低了
> 
> 我本地的cuda版本是12.2。不知道是不是镜像里的显卡驱动有问题？如果方便的话，能提供下dockerfile不?

请问这个问题你解决了嘛？我也遇到了这个问题

sybarite666, 2024-10-09:
请问这个问题你解决了嘛？我也遇到了相同的问题