w111liang222/lidar-slam-detection #37: No directory showed in the page: preview

Fernweh-yang, 2024-06-17:
Hi, I want to follow the the [doc](https://github.com/w111liang222/lidar-slam-detection/blob/main/docs/slam.md#%E5%9C%B0%E5%9B%BE%E7%BC%96%E8%BE%91) to open one pcd file.  
But as the figure showed, it is blank:
[image]

w111liang222, 2024-06-17:
lsd的地图格式不是pcd，是关键帧+g2o节点边，你可以先使用demo数据跑一下，保存到磁盘，然后看一下具体格式

Fernweh-yang, 2024-06-17:
> lsd的地图格式不是pcd，是关键帧+g2o节点边，你可以先使用demo数据跑一下，保存到磁盘，然后看一下具体格式

但在preview中貌似让我选g2o文件的地方也木有T T

w111liang222, 2024-06-17:
preview是选回放数据的，不是地图，选地图在/editor 页面下，参考这个文档https://github.com/w111liang222/lidar-slam-detection/blob/main/docs/guide.md
有挂载虚拟磁盘，然后/root/exchange/lp_log目录下放置demo文件夹，即可在preview页面下看到了