w111liang222/lidar-slam-detection #38: No sensor config as the doc showed

Fernweh-yang, 2024-06-17:
Hi, I want to follow the [doc](https://github.com/w111liang222/lidar-slam-detection/blob/main/docs/slam.md#%E4%BD%BF%E7%94%A8%E6%A1%88%E4%BE%8B) to run slam with the demo dataset. But in sensor configuration there're no camera and rs-lidar-16:
[image]
By the way, where can I set the dataset path?

w111liang222, 2024-06-17:
cfg/board_cfg_all.yaml文件中的input.data_path可以指定运行的数据

Fernweh-yang, 2024-06-17:
> cfg/board_cfg_all.yaml文件中的input.data_path可以指定运行的数据

设置了cfg/board_cfg_all.yaml文件中的input.data_path后，config页面可以选择相机和雷达了。但是我点击页面右下角的上传按钮(保存设置？)后，貌似就卡死在这个页面了，这里可能是什么原因呢？
[image]

w111liang222, 2024-06-17:
麻烦提供下完整的log，我分析一下