w111liang222/lidar-slam-detection #41: web预览camera窗口闪烁

zhu2900000, 2024-06-20:
你好楼主，web窗口camera显示不稳定，我在程序内部获取是正常的如print（sensor.read()），我的窗口大小是1920 1080的，我调整了640 480 仍然有这个问题，我怀疑有可能是web_ui的问题。

w111liang222, 2024-06-20:
你的相机帧率是不是太低了？要大于激光帧率才不会闪

zhu2900000, 2024-06-20:
> 你的相机帧率是不是太低了？要大于激光帧率才不会闪
相机是25fps，雷达是10hz， 不是内容闪烁，是断开连接造成的闪烁，大概3s断一次，1s后又恢复。

w111liang222, 2024-06-20:
把激光先关掉，看一下preview页面的右下角FPS

w111liang222, 2024-06-20:
传感器优先级是"激光” > “相机”
激光20Hz，相机15hz，在拿到一帧激光后，如果没有相机数据，就会闪
当关闭激光雷达后，则相机优先级最高，按相机的帧率去获取数据，不会闪

zhu2900000, 2024-06-20:
我现在把相机帧率提高到25fps，激光的不变，还是闪。。。。。。

zhu2900000, 2024-06-20:
> 传感器优先级是"激光” > “相机” 激光20Hz，相机15hz，在拿到一帧激光后，如果没有相机数据，就会闪 当关闭激光雷达后，则相机优先级最高，按相机的帧率去获取数据，不会闪

单独了聊一下这个，如果queue中相机没有更新就保留上一帧的，这样更合理把哈

zhu2900000, 2024-06-20:
我把lidar 将到600rpm 10hz 解决了

w111liang222, 2024-06-20:
> > 传感器优先级是"激光” > “相机” 激光20Hz，相机15hz，在拿到一帧激光后，如果没有相机数据，就会闪 当关闭激光雷达后，则相机优先级最高，按相机的帧率去获取数据，不会闪
> 
> 单独了聊一下这个，如果queue中相机没有更新就保留上一帧的，这样更合理把哈 我原来做过，多传感器无关频率，每个传感器都能取道各自最新的数据。

看需求的，如果传递重复数据给后续模块，就得判断是否是重复数据

zhu2900000, 2024-06-20:
@w111liang222 都是最新的就行。可以用时间戳做延迟补偿

w111liang222, 2024-06-20:
https://github.com/w111liang222/lidar-slam-detection/blob/959ad59ce01f08feeba70404d3a37e36c0328a89/module/source/camera_data_manager.py#L271

修改该行代码，timeout=1.0 就是你说的这种模式