w111liang222/lidar-slam-detection #45: 系统无法完成初始化，module_manager.py :113 timeout: system initializing error

sybarite666, 2024-09-27:
4070TI, 设备状态一直卡在初始化中，想请教一下如何解决
[image]
[image]
[image]

w111liang222, 2024-09-27:
检查一下程序启动开始有没有报错？

sybarite666, 2024-09-30:
前天使用时，系统启动开始没有报错，一直都是绿色的正常运行。
直到这一行出现了module_manager.py :113 timeout: system initializing error 
昨天运行时，又出现了新的问题：CUDA faliure: CuDA driver is insufficient for CUDA runtime version
我的GCCversion是9.4.0

[image]
[image]