w111liang222/lidar-slam-detection #51: FastLio version ans other algorithms

ilay-chen, 2024-12-29:
Hi!
thank you for this amazing repo!
What version of FastLio do you use? 1? 2?
Is it possible to upgrade? And how complicated is it to add new algorithms (like fast_livo etc)

Thank you!

w111liang222, 2024-12-30:
> Hi! thank you for this amazing repo! What version of FastLio do you use? 1? 2? Is it possible to upgrade? And how complicated is it to add new algorithms (like fast_livo etc)
> 
> And also, all the provided algorithms: "[lidar_align](https://github.com/ethz-asl/lidar_align): A simple method for finding the extrinsic calibration between a 3D lidar and a 6-dof pose sensor. [lidar_imu_calib](https://github.com/chennuo0125-HIT/lidar_imu_calib): automatic calibration of 3D lidar and IMU extrinsics. [OpenPCDet](https://github.com/open-mmlab/OpenPCDet): OpenPCDet Toolbox for LiDAR-based 3D Object Detection. [AB3DMOT](https://github.com/xinshuoweng/AB3DMOT): 3D Multi-Object Tracking: A Baseline and New Evaluation Metrics. [FAST-LIO](https://github.com/hku-mars/FAST_LIO): A computationally efficient and robust LiDAR-inertial odometry package. [R3LIVE](https://github.com/hku-mars/r3live): A Robust, Real-time, RGB-colored, LiDAR-Inertial-Visual tightly-coupled state Estimation and mapping package. [FLOAM](https://github.com/wh200720041/floam): Fast and Optimized Lidar Odometry And Mapping for indoor/outdoor localization. [hdl_graph_slam](https://github.com/koide3/hdl_graph_slam): an open source ROS package for real-time 6DOF SLAM using a 3D LIDAR. [hdl_localization](https://github.com/koide3/hdl_localization): Real-time 3D localization using a (velodyne) 3D LIDAR. [ORB_SLAM2](https://github.com/raulmur/ORB_SLAM2): Real-Time SLAM for Monocular, Stereo and RGB-D Cameras, with Loop Detection and Relocalization Capabilities. [scancontext](https://github.com/irapkaist/scancontext): Global LiDAR descriptor for place recognition and long-term localization. " Do you use them directly, or do you alter them to your specific version? I ask to understand the flexibility...
> 
> Thank you!

Hi,
Thank you for your attention，we use FASTLIO2 as the frontend lidar odometry, if you want to add new odometry, the abstract interface of 'SlamBase' class should be implemented.

These referenced repos are modified to adapted to LSD.

ilay-chen, 2024-12-30:
Hi,
Thank you for the quick response!

I would be happy if you could assist me with these questions also:
1. What is the difference between the GPU and the non-GPU docker? Only the detection part? Or is there more difference?
2. Mapping is basically SLAM and localization is only if we already have a map?
3. Mapping is real-time? I saw in in GUI that the location is moving in 10HZ, but the trajectory line (green line with connected dots), is in lower frequency. What is its meaning?
[image]
The green trajectory is in low frequency. Why?
4. I see only two options in SLAM option. There is more?
5. How to change SLAM parameters? I see only a few options to change...
In this screen, I see only two options and a few SLAM configurations (questions 4, 5):
[image]

Thank you very much!

w111liang222, 2024-12-31:
Hi,

1. The difference between GPU and non-GPU docker is that the CUDA will be checked to decide if enable the **NDT_CUDA** for map matching and loop closure detection.
2. Yes, for autonomous vehicle driving scenario, we normally build a prior pointcloud map by mapping (SLAM) and then the map will be used for annotation or localization.
3. Mapping is real-time, the location of LiDAR odometry is 10Hz updated in GUI. The green line means the keyframe poses in PGO and updated every 3s.
4. 5. We only export the very important parameters to user for simplified usage. The current setting of algorithm can cover most cases( based on our extensive tests). Also you can modify the source code to expose more parameters.

You are welcome.

w111liang222, 2025-01-09:
If you want to test the LSD with rosbag, you need to follow the steps strictly:
1. launch the rosbag_proxy node firstly
2. then run the LSD to receive the camera data. due to the rosbag_node acts as a http server, it should setup before client (LSD process)

I recommend to test LSD with public dataset which is already converted to .pkl files. and you can also convert your own bag to pkl by the *rosbag_to_pkl* tool

ilay-chen, 2025-01-09:
Thank you!
I cant see the images is sending at all...
this is the cpp of the camera proxy:
#include "web_video_server/web_video_server.h"

#include "ros/ros.h"

int main(int argc, char **argv) {
    ros::init(argc, argv, "rosbag_camera_proxy");

    ros::NodeHandle nh;
    ros::NodeHandle private_nh("~");
    web_video_server::WebVideoServer server(nh, private_nh);
    server.spin();
    return 0;
}

its looks empty?

Also, I want to check the system live. This is why I want with rosabag..
there is a realtime "production-like" version of running?

thanks a lot!

w111liang222, 2025-01-10:
Hi, ilay-chen

The the implementation of camera proxy is in the src/web_video_server.cpp

Sorry, I don't know the exact mean of "realtime "production-like" version", The LSD can directly connect to the LiDAR and camera sensors. and many http APIs are exported (for example: POST /v1/status, can help to check the running status of the system)