w111liang222/lidar-slam-detection #54: 没找到地图合并的功能

nieyan, 2025-01-25:
使用docker的纯cpu环境启动，参考文档： https://github.com/w111liang222/lidar-slam-detection/blob/main/docs/slam.md 
正常建图完成，进editor界面打开了上述建好的地图；
并copy了一份在 /media/lp_log/map目录下，想使用两份map测试地图合并功能，但是网页上好像没有这个选项？

[image]

w111liang222, 2025-01-25:
> 使用docker的纯cpu环境启动，参考文档： https://github.com/w111liang222/lidar-slam-detection/blob/main/docs/slam.md 正常建图完成，进editor界面打开了上述建好的地图； 并copy了一份在 /media/lp_log/map目录下，想使用两份map测试地图合并功能，但是网页上好像没有这个选项？
> 
> [image]

设置slam模式为localization，才会出现“合并”选项

nieyan, 2025-01-25:
> > 使用docker的纯cpu环境启动，参考文档： https://github.com/w111liang222/lidar-slam-detection/blob/main/docs/slam.md 正常建图完成，进editor界面打开了上述建好的地图； 并copy了一份在 /media/lp_log/map目录下，想使用两份map测试地图合并功能，但是网页上好像没有这个选项？
> > [image]
> 
> 设置slam模式为localization，才会出现“合并”选项

合并后，点击保存地图，网页卡住了，看log好像保存失败了
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/x86_64-linux-gnu/libthread_db.so.1".
30      ../sysdeps/unix/sysv/linux/epoll_wait.c: No such file or directory.
0x00007fdc7bcb746e in epoll_wait (epfd=50, events=0x70ce810, maxevents=64, timeout=1000) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30

Thread 32 (Thread 0x7fdbe2ffd700 (LWP 3109)):
#0  0x00007fdc7bc7ac7f in __GI___wait4 (pid=3110, stat_loc=0x0, options=0, usage=0x0) at ../sysdeps/unix/sysv/linux/wait4.c:27
#1  0x00007fdc75bc0c03 in print_trace() () at /home/znqc/work/lidar-slam-detection/build/lib.linux-x86_64-3.8/libcpp_utils.so
#2  0x00007fdc75bc0c1c in signal_handler(int) () at /home/znqc/work/lidar-slam-detection/build/lib.linux-x86_64-3.8/libcpp_utils.so
#3  0x00007fdc7bbdb090 in <signal handler called> () at /usr/lib/x86_64-linux-gnu/libc.so.6
#4  0x00007fdc6223ac97 in g2o::save_robust_kernels(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, g2o::SparseOptimizer*) () at /home/znqc/work/lidar-slam-detection/build/lib.linux-x86_64-3.8/libg2o_backend.so
#5  0x00007fdc62213aad in hdl_graph_slam::GraphSLAM::save(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () at /home/znqc/work/lidar-slam-detection/build/lib.linux-x86_64-3.8/libg2o_backend.so
#6  0x00007fdc621d32c9 in graph_save(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () at /home/znqc/work/lidar-slam-detection/build/lib.linux-x86_64-3.8/libg2o_backend.so
#7  0x00007fdc629d7c70 in  () at /usr/local/lib/python3.8/dist-packages/perception-1.0.0-py3.8-linux-x86_64.egg/slam_wrapper.so
#8  0x00007fdc629e7a2d in  () at /usr/local/lib/python3.8/dist-packages/perception-1.0.0-py3.8-linux-x86_64.egg/slam_wrapper.so
#9  0x00007fdc62a01046 in  () at /usr/local/lib/python3.8/dist-packages/perception-1.0.0-py3.8-linux-x86_64.egg/slam_wrapper.so
#10 0x00000000005f6929 in PyCFunction_Call ()
#11 0x00000000005f74f6 in _PyObject_MakeTpCall ()
#12 0x0000000000571164 in _PyEval_EvalFrameDefault ()
#13 0x00000000005f6cd6 in _PyFunction_Vectorcall ()
#14 0x000000000050c149 in  ()
#15 0x00000000005f6082 in PyObject_Call ()
#16 0x000000000056d2d5 in _PyEval_EvalFrameDefault ()
#17 0x00000000005f6cd6 in _PyFunction_Vectorcall ()
#18 0x000000000056bbfa in _PyEval_EvalFrameDefault ()
#19 0x00000000005f6cd6 in _PyFunction_Vectorcall ()
#20 0x000000000056bbfa in _PyEval_EvalFrameDefault ()
#21 0x00000000005f6cd6 in _PyFunction_Vectorcall ()
#22 0x000000000050bc2c in  ()
#23 0x00000000005f6082 in PyObject_Call ()

nieyan, 2025-01-25:
另外，多趟合并除了提高点云的稠密度，能提高建图精度不？

w111liang222, 2025-01-25:
> 另外，多趟合并除了提高点云的稠密度，能提高建图精度不？
合并时会对每个关键帧附近搜索可能存在的回环，某些场景下能提高建图精度

nieyan, 2025-01-25:
> > 另外，多趟合并除了提高点云的稠密度，能提高建图精度不？
> 
> 合并时会对每个关键帧附近搜索可能存在的回环，某些场景下能提高建图精度

好的；另外上述合并后存图报错的问题，是否和地图参数-距离步长不同有关？ 同一批数据，一个步长1m， 一个步长0.1m， 但我看合并是没报错来着；  另外 点云着色相关的代码 slam/vio 没找着呢？

w111liang222, 2025-01-26:
> > > 另外，多趟合并除了提高点云的稠密度，能提高建图精度不？
> > 
> > 
> > 合并时会对每个关键帧附近搜索可能存在的回环，某些场景下能提高建图精度
> 
> 好的；另外上述合并后存图报错的问题，是否和地图参数-距离步长不同有关？ 同一批数据，一个步长1m， 一个步长0.1m， 但我看合并是没报错来着； 另外 点云着色相关的代码 slam/vio 没找着呢？

不太清楚为啥报错，用其他数据测试看看？点云着色在slam/localization/下

nieyan, 2025-02-06:
> > > > 另外，多趟合并除了提高点云的稠密度，能提高建图精度不？
> > > 
> > > 
> > > 合并时会对每个关键帧附近搜索可能存在的回环，某些场景下能提高建图精度
> > 
> > 
> > 好的；另外上述合并后存图报错的问题，是否和地图参数-距离步长不同有关？ 同一批数据，一个步长1m， 一个步长0.1m， 但我看合并是没报错来着； 另外 点云着色相关的代码 slam/vio 没找着呢？
> 
> 不太清楚为啥报错，用其他数据测试看看？点云着色在slam/localization/map_render

点云着色这块能否细化下操作步骤， 感恩~

nieyan, 2025-02-06:
点云着色功能如何启动？
尝试使用建图功能生成了地图并保存在lp_log/map/test1目录下，在配置中切换到定位模式，开启点云着色，到editor界面没看到变化
[image]

nieyan, 2025-02-07:
假设重建好一份地图，想在该地图中获取第K帧时刻的camera位姿，从而生成occ真值；
该camera位姿是使用
1. 重建的lidar轨迹 -》K帧lidar轨迹 -》标定好的lidar和camera外参换算 -》K帧camera位姿；
2. K帧点云重定位  -》点云着色中提及的步骤优化获得K帧camera位姿
3. 其他方法？

哪种方式生成的occ真值更科学呢？
如果是使用类似fast-LIVO2方式重建的地图，是否直接取重建的轨迹中的k帧camera位姿即可（建图优化考虑了重投影误差）;
在fast-LIVO2等方法中，是否默认lidar和camera的外参是一个刚性连接，不会变化？那重投影误差影响的只是机器人的位姿吗？

w111liang222, 2025-02-07:
我这边经验是，标定分开做，放在滤波器中实时估计雷达-相机外参，可能会有误差比较大的情况

nieyan, 2025-02-07:
好的！雷达-相机外参这块理解了。
关于occ真值这块，是不是拆分成两个指标去评估:
1. 正确的机器人位姿    2. 上述机器人位姿下的正确的camera位姿（重投影误差越小越nice）

方式2 点云重定位的 机器人位姿相较于方式1是更准确的（无累计误差）；在此基础上再优化camera位姿，最后得到的occ真值是更科学的

nieyan, 2025-02-13:
[image]
想问下slam_mapping_demo的pkl文件里存的image是做过去畸变的吗？ 我把对应的单帧lidar投影回图像，感觉结果不太对

w111liang222, 2025-02-13:
没有去畸变

nieyan, 2025-02-14:
> 没有去畸变

cfg.yaml 里的 extrinsic是 camera2lidar   or lidar2camera呢？

w111liang222, 2025-02-14:
lidar to camera