w111liang222/lidar-slam-detection #57: 如何查看点云着色的效果？

Harry-675, 2025-03-12:
[image]请问一下点云着色的程序是没有执行嘛，点云显示选择颜色后会什么都不显示。

w111liang222, 2025-03-12:
Hi,
目前点云着色已经修改为通过tools/postprocessing/colouration_cloud.py脚本对点云着色

w111liang222, 2025-03-12:
> > Hi, 目前点云着色已经修改为通过tools/postprocessing/colouration_cloud.py脚本对点云着色
> 
> 感谢回复。另外有个问题，看colouration_cloud.py是对pkl中的点云进行投影着色，这里会进行点云去畸变操作嘛，另外读取点云的时候要读取points_attr参数，但是demo数据里并没有这个字段。

会做去畸变
demo数据比较老了，points_attr里边存的是每个点的时间戳和ring等信息，这个是后来版本加上的

Harry-675, 2025-03-13:
> > > Hi, 目前点云着色已经修改为通过tools/postprocessing/colouration_cloud.py脚本对点云着色
> > 
> > 
> > 感谢回复。另外有个问题，看colouration_cloud.py是对pkl中的点云进行投影着色，这里会进行点云去畸变操作嘛，另外读取点云的时候要读取points_attr参数，但是demo数据里并没有这个字段。
> 
> 会做去畸变 demo数据比较老了，points_attr里边存的是每个点的时间戳和ring等信息，这个是后来版本加上的

请问有新的示例数据嘛，不知道该怎么组织自己的数据

w111liang222, 2025-03-13:
可以参考这个工具: **tools/rosbag_to_pkl**，可以把其他数据转换到LSD格式

sakura-yy520, 2025-06-04:
请问点云着色需要gpu吗