w111liang222/lidar-slam-detection #58: Can I know on which datasets your CenterPoint model has been trained?

Zhao-Qihao, 2025-03-18:
Great work! Can I know on which datasets your CenterPoint model has been trained?

w111liang222, 2025-03-18:
> Great work! Can I know on which datasets your CenterPoint model has been trained?

Hi,
The nuscense, WOD and Lyft datasets are combined to train the center point model

Zhao-Qihao, 2025-03-22:
Hi, I try the same method to concat the there dataset, but the problem is that waymo dataset dose not have the class 'traffic corn', how did you do combine the there dataset to train your model

w111liang222, 2025-03-22:
> Hi, I try the same method to concat the there dataset, but the problem is that waymo dataset dose not have the class 'traffic corn', how did you do combine the there dataset to train your model

you can use class mask for each dataset